miguelbalboa/rfid #66: read card once only

e4awais, 2015-01-07:
hi, 

i am doing a project related to rfid, and am taking help from the library u shared. thanks :) 

as i scan and read an authenticated card, it does so and if the card is kept placed at the reader, it reads it only once.

but as i scan an unauthenticated card, it keeps scanning if kept placed at the reader, i need it to scan only once, notify the not authentication part and leave it as thus until a NEW card is with in the proximity. 

i am calling the halt and stop cryptol functions as should be called. 

thanks

e4awais, 2015-01-07:
there is another issue, when i keep scanning cards, after sometime read failure starts happening and status has no room aka buffer is not big enough starts getting printed in the debugger. 

thanks

e4awais, 2015-01-07:
i tried to debug the above issue, it happens when "the CRC_A does not match" occurs. now when ever status_no_room or status_time_out etc occur, i m able to redo them via a software reset using .PCD_Reset() ... but i doesnt work when crc error occurs ... please help ... 

please not that at present i m not using the (hardware ) reset pin of mfrc522 module. 

appreciate all. 

thanks

StanTheMan132, 2017-08-17:
Not sure if we are having the same issue but it sounds similar. It works fine the first time but once i place the same or different tag it doesn't read it anymore and i get no message in serial as if there is nothing there.  I found out this is because of something when authenticating, not sure what. Note even when then running a other program I still cannot read from the Rfid reader until i have turned the Arduino on and off. Help Appreciated!

AyushMarsian, 2018-04-19:
@StanTheMan132 @e4awais 

Please [visit following page](https://github.com/miguelbalboa/rfid/issues/323)  may be this conversation will helpful to you.

wanoo, 2018-08-17:
Hello,
I have the same issue, if i wave a card up and down over and over i will et a CRC_A error, after that the reader is jam and i need to restart the arduino to get it work again, the .PCD_Reset() don't fixit.

Thanks

porohkun, 2018-12-24:
hi

have the same or similar issue. I cant auth and read one sector twice without removing card from reader and placing it again.

reproduction code: https://pastebin.com/i9WmXiCC (skip to line 176 for loop() )

stephanbrunner, 2019-05-06:
hi @porohkun 
Have you solved your issue? I'm having the same problem. ReqA -> Select- > Auth -> Read works, but if I try another Auth -> Read, I get athentication failure: Timeout in communication.

I had a workaround with antennaOff -> antennaOn, but then run in other prolems later, so I would love to have it the right way around to avoid followup issues.

porohkun, 2019-05-06:
hi @stephanbrunner 
Nope. I have change reader to pn532 and use other lib for longer reading distance. But this issue was away too.

stephanbrunner, 2019-05-06:
Ok. My issue seems to be fixed. I called StopCrypto before auth for another block. But StopCrypto should only be called before using another PICC.