miguelbalboa/rfid #67: I uploaded this code on my arduino uno. But it does not seem to work. I am getting a NACK from the PICC. What changes should I do?

RitikaMalik, 2015-01-17:


omersiar, 2015-01-20:
Which code did you upload? Do you have another PICC?

RitikaMalik, 2015-01-20:
I uploaded the Read/Write Code that is a part of the attached examples file. Yes, I do have 2 PICCs. I am facing the same problem in both. 
One more thing, I looked into the .c file of the MFRC522 library where I stumbled upon a function called "DumpMifareClassicSectorToSerial". This function works fine and it internally calls MIFARE_Read(). This means that the read function is also but the way in which has been called in that Read/Write code is wrong.
Am I thinking in the right direction?
Thanks for the help.

omersiar, 2015-01-20:
Reading/Writing requires Authentication which is not always done by Library examples default (different keys). You need to modify for your needs.

RitikaMalik, 2015-01-20:
I do agree to the authentication part. But if the authentication is working fine in "DumpMifareClassicSectorToSerial", then why does it fail in the example code?

RitikaMalik, 2015-01-20:
And I will go through the functions once again, may be I lack some basic understanding of the same.
Thanks.

mdxs, 2015-03-16:
@RitikaMalik : Is this still an issue for you?  Otherwise, please report back so that this issue can be closed. In general I would suggest to first run the DumpInfo example before trying ReadAndWrite (and note that ReadAndWrite will attempt to change data... which, as @omersiar said may require authentication).

RitikaMalik, 2015-03-16:
No, this is no more an issue for me. :)

gnrcampos, 2015-03-30:
I think I am having an authentication issue here, how am I supposed to know the authentication key for my tags and cards?

mdxs, 2015-03-30:
I think this issue should be closed.

@gnrcampos When you use authentication on your own (originally) blank tags/cards, you should remember it yourself. When that fails, on blank tags/cards, perhaps the "rfid_default_kes" in examples (and the link in the comments) can help you. When you try to access tags/cards that are controlled by others (like public library, transportation, bank, etc.)... then you are perhaps not supposed to know the authentication keys ;-)

closed 2015-06-16 by miguelbalboa