miguelbalboa/rfid #75: ReadAndWrite: insufficient memory

skurella, 2015-01-29:
When trying to compile ReadAndWrite example for Arduino Pro Mini with Atmega328 (8Mhz, 3.3V) the following error appears:
`Sketch uses 10,688 bytes (34%) of program storage space. Maximum is 30,720 bytes.
Global variables use 2,078 bytes (101%) of dynamic memory, leaving -30 bytes for local variables. Maximum is 2,048 bytes.
processing.app.debug.RunnerException: Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
	at processing.app.Sketch.size(Sketch.java:1679)
	at processing.app.Sketch.build(Sketch.java:1589)
	at processing.app.Sketch.build(Sketch.java:1508)
	at processing.app.Editor$DefaultRunHandler.run(Editor.java:1912)
	at java.lang.Thread.run(Unknown Source)
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.`
N.B. this only happens on Arduino 1.5.8. On 1.0.6 code compiles but does not work.
N.B.2 Library version from 26 Oct 2014 works, with stability warning from the compiler.

skurella, 2015-01-29:
Found the problem. Error strings in the library take up most of the available dynamic memory on 2KB microcontrollers. I suggest moving the strings to flash memory (PROGMEM).

miguelbalboa, 2015-01-29:
Hello @skurella ¿Can you explain what did you do or make a pull? 
Greetings!

lorneb, 2015-01-30:
Hi i'll do a pull request now,  I've made the changes to move the strings to program mem using 
Serial.print(F("The literal String"))

lorneb, 2015-01-30:
Arrhh,  git on my Mac has capped out, will push the changes tomorrow cheers

miguelbalboa, 2015-01-30:
See 605a1d755f3762e037cb7545c5cd868794d66ce7,
Thanks to @skurella

closed 2015-01-30 by miguelbalboa