miguelbalboa/rfid #84: A MIFARE PICC responded with NAK.

Grbaset, 2015-02-19:
I tried to use your example ReadAndWrite, I customized data and it worked, but now it says: MIFARE_Write() failed: A MIFARE PICC responded with NAK. It's a different data and card, but the code is the same.

mdxs, 2015-03-16:
So it works for one PICC and not for another, perhaps authentication?

Grbaset, 2015-03-17:
No. It works only with one. The 3 that are not it doesn't work. And they are the same type.

Mohamad1994HD, 2015-12-26:
MIFARE_Write() failed: A MIFARE PICC responded with NAK.

I have the same problem, even I have not changed anything in the code.

Nurza, 2016-06-24:
Same problem here, up ?

Basti86, 2016-06-27:
Try to authenticate with:
`Serial.println("Authenticating again using key B...");
status = mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, trailerBlock, &key, &(mfrc522.uid));`
instead of:
`Serial.println("Authenticating again using key B...");
status = mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_B, trailerBlock, &key, &(mfrc522.uid));`
Per default some cards are delivered with the access bits set in a way that KeyB can be read from the card. Therefore you can not write with KeyB and therefore you have to use keyA. (http://www.mouser.com/ds/2/302/MF1S503x-89574.pdf Section 8.7.1 - 8.7.3)

Grbaset, 2016-06-28:
Thanks! It works.

closed 2016-08-25 by Rotzbua