miguelbalboa/rfid #87: ARDUINO MEGA + Ethernet Shield + RFID-RC522

EscuderoKevin, 2015-02-24:
Good night all.

Well I came a rush job to a factory where you have to make a control system USERS controlled via web, etc etc.

But hey need to start with the part of the RFID and sincerely and booting problems.

I commented.

Use thie code DUMPInfo

Conex Images: https://www.dropbox.com/sh/whjm71iaua9luzt/AABp8Vf3b9Fj9ToyvUwNDn2ra?dl=0

= 

ERRORS:

MFRC522 Software Version: 0x91 = v1.0
Scan PICC to see UID, type, and data blocks...
Card UID: 43 B5 45 DF
PICC type: MIFARE 1KB
Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits
  15     63  MIFARE_Read() failed: The CRC_A does not match.
  15     62   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ]  Inverted access bits did not match! 
         61  MIFARE_Read() failed: A MIFARE PICC responded with NAK.
         60  MIFARE_Read() failed: The CRC_A does not match.
  14     59  PCD_Authenticate() failed: Timeout in communication.
  13     55  PCD_Authenticate() failed: Timeout in communication.
  12     51  PCD_Authenticate() failed: Timeout in communication.
  11     47  PCD_Authenticate() failed: Timeout in communication.
  10     43  PCD_Authenticate() failed: Timeout in communication.
   9     39  PCD_Authenticate() failed: Timeout in communication.
   8     35  PCD_Authenticate() failed: Timeout in communication.
   7     31  PCD_Authenticate() failed: Timeout in communication.
   6     27  PCD_Authenticate() failed: Timeout in communication.
   5     23  PCD_Authenticate() failed: Timeout in communication.
   4     19  PCD_Authenticate() failed: Timeout in communication.
   3     15  PCD_Authenticate() failed: Timeout in communication.
   2     11  PCD_Authenticate() failed: Timeout in communication.
   1      7  PCD_Authenticate() failed: Timeout in communication.
   0      3  PCD_Authenticate() failed: Timeout in communication.

mdxs, 2015-03-16:
Perhaps the PICC (the card that you present to the reader) needs authentication to read specific data (I think the "Inverted access bits did not match!" message hints at that...) or there is a Cyclic Redundancy Check (CRC) error with some data/access block. I'm no expert (just played with Arduino Uno and RFID reader) but from what I read, there might be cases that a PICC is "damaged" when pulled during a write cycle... but I have no idea if it would present itself like this.

You didn't describe whether you have had any success with other PICCs that are known to work.

closed 2015-12-05 by miguelbalboa

jhed20, 2016-01-24:
I also encountered that problems.. Please help..