miguelbalboa/rfid #88: RFID RC533 and ISO 14443-4 transmission protocol

Sztef89, 2015-03-01:
Hi!

Thanks a lot for the library!
I would like to know if there is a possibility to add support for 14443-4 (transmission protocol). I would like to exchange data between my phone and my other cards of this type. I see this for the functionality is not added.

Is it possible to add support for this standard and if you have it on your TODO list?
After reading the documentation (look at image) I came to the conclusion that my knowledge on this topic is too low :/

[image]

Thank you in advance!

--
Best regards
Stefan

mdxs, 2015-03-16:
@Sztef89 : it is unclear to me what you have tried, what worked, and what you want to try and accomplish with this library. Should we assume you have an Arduino and an RFID RC522 based module (given the picture you do)?  If so, try the DumpInfo example first.

Sztef89, 2015-03-19:
I trying to read and write data from card and phone (galaxy S4) using "14443-4 Transmission protocol".

Dump info:
MFRC522 Software Version: 0x92 = v2.0
Scan PICC to see UID, type, and data blocks...
Card UID: [CENSORED]
PICC type: PICC compliant with ISO/IEC 14443-4
Dumping memory contents not implemented for that PICC type.

I see this function isn't implemented but is possible to transfer data with this module RC522?

How hard for you is make this implementation? :)

mdxs, 2015-03-30:
As hard as it is for you... or actually harder (as others may not have such PICCs/tags/cards). So you are welcome to provide a Pull Request that implements this.

Unless you are going to do this, I suggest you close this issue: it's "not implemented for that PICC type" is the best the current implementation of the library will do.

Rotzbua, 2015-04-01:
Maybe you can use some implementations from the PN532:
https://github.com/Seeed-Studio/PN532

Sztef89, 2015-04-01:
My knowledge is to low to do it but thx for link :( I trying some comunication with phone but is impossible with this library. Maybe we found in future some peoples who do it this implementation. I hope so :)

closed 2015-12-05 by miguelbalboa

oilol, 2017-01-30:
http://stackoverflow.com/questions/41937736/rc522-nothing-received-in-sl3-mifare-card
Hello, can anyone help me?

knopserl, 2019-03-22:
I have also many of this RC522 cards with firmware v92 version 2 which is the most recent version from NXP.
THe library is really great, but I have now new Security Chips (Cards) with obviously ISO 14443-4 type and cant communicate with the controller. I guess the same type of chips on my Maeastro and Credit Card as I have the same type and the same problem.
I have used the MFRC522Extended.h class  
I get timeouts if I communicate with the cards using the proper (according  to the docu) command structure.
I have used both: TCL_Transceive and PCD_TransceiveData.
I think it might be also a problem of the MFRC522 readers as they could have problems with 
ISO 14443-4 (non A) cards. Can anybody confirm that or have a solution?
I have already odered otehr readers like PN532 which claims to read ISO 14443-4 cards.

Here my log:

UID: 0123456789abcdefaa55
Card UID: 01 23 45 67 89 AB CD EF AA 55
Card SAK: 20
PICC type: PICC compliant with ISO/IEC 14443-4
Dumping memory contents not implemented for that PICC type.
 
PCD_TransceiveData status=3
0000000000000000000000000000000000000000000000000000000000000000