miguelbalboa/rfid #92: RFID interrupt pin IRQ setup problem

Nilsom, 2015-03-22:
How do I set up the IRQ-pin on the board with Arduino? I want the IRQ-pin to change value when a card is scanned on my rc522. Code-examples would be appreciated! Sorry for my English…

aki263, 2015-03-29:
I am also having same issue. IRQ is not working

gnrcampos, 2015-03-30:
Also need that for my project... Have already tried some code but the datasheet says too little about it..

Rotzbua, 2015-03-31:
It seems that the irq pin is deactivated by default.
I looked into the data sheet and I think you need to set bits in register ComIEnReg. Page 39 table 26 says something how the irq pin can be triggered and which events can generate a signal.
But I do not know which events/bits are useful.

gnrcampos, 2015-03-31:
I've set those bits but neither one seems to be related to the presence of the card...

t3chguy, 2015-10-07:
I too would really appreciate an example showing how to enable the IRQ output on RC522s

oozel, 2015-12-02:
Interrupt functions are controlled with DivIEnReg, DivIrqReg, ComIEnReg, ComIrqReg registers. But there is no any example or detailed explanation about that.

closed 2015-12-05 by miguelbalboa

t3chguy, 2015-12-20:
I know that I never managed to get it working, I tried for a few hours and my scope was able to trigger on the signal from IRQ but it looked to be such a minute change that an MCU would never be able to trigger on it. Would really appreciate someone sharing how to get it working as IRQ would make this peripheral much more handy.

Thanks

jphgross, 2016-02-03:
After spending a day experimenting and mulling over the spec, I believe that the interrupt behavior is not there to indicate any card is in the field but instead to assist with communications with the device/card. The drivers that I looked at poll on specific bits that indicate either the communication completed, the buffer is almost full, or almost empty, timed out, errored out, etc. IE. when sending or receiving data from the card the driver must check the status, assuming it would work properly one could wait on a conditional that was set by the interrupt handler instead of polling in a tight loop.

I was looking to see if there might be some trick that could be played, but I'm at the end and am conceding to polling on the interface.

I know that the pin is actually connected and will go high/low. Setting it to DivIEnReg to 0x84 (set the pin to CMOS push/pull and enable DivIEnReg had an effect, but its not related to a card in the field and resets at a regular interval. 

It's too bad since its chewing up alot cycles checking on my system.

masterleo, 2017-04-18:
Hello I have a big problem, I need to connect severals RFID modules on the same arduino, how to do it ????

albertoVGitHub, 2018-02-03:
Hello all
Somebody has already figured out this issue?
I would appreciate any help :)

Rotzbua, 2018-02-03:
As mentioned in the README there is an example how you can use/enable irq. The library it self do not use irq. The irq can be used to interact with the rfid send/receive queue (and some other events), but nothing really useful. The lib just use a easy pull mechanism for send and receive.

@albertoVGitHub What issue? I think you didn't read the thread?