miguelbalboa/rfid #93: Port for Linkit One

aki263, 2015-03-26:
I tried every possible way to connect wires. But its not recognising card, i get this msg in few sec http://prntscr.com/6lhdi8 but card is not being detected.

Rotzbua, 2015-03-26:
I bought a set from a Chinese seller. The token works well, but the card is not detected.
Recording to some other comments, it seems to be problem for more people. But they did not mention the problem or a solution.

I solved it by only using tokens ;)
Note: if you buy them from Chinese seller, there could be some dead tokens

aki263, 2015-03-27:
My reader is working perfect and tokens are also good beause i tested both in raspberryPi. At the time when i was writing the issue i had not find the fix but now i have a ino that reads my card http://www.grantgibson.co.uk/blog/wp-content/uploads/2012/04/ggrfid_en_ino.txt

Now i wonder why this library dont work.

Rotzbua, 2015-04-01:
Thanks for the ino. But my card does not work with both codes. (I think the card is dead; tested with a arduino uno)

I found a difference at the init code timer settings. This bib uses a different values than normally.

Actual bib:
```
(TModeReg, 0x80);
(TPrescalerReg, 0xA9);
(TReloadRegH, 0x03);
(TReloadRegL, 0xE8);
```
found at raspberry code and the ino file:
```
(TModeReg, 0x8D);
(TPrescalerReg, 0x3E);
(TReloadRegH, 0);
(TReloadRegL, 30);
```

Maybe a change helps?

Rotzbua, 2016-08-25:
@aki263 Hi. Did you found a solution? Or a more detailed bug description?
Otherwise I will close this issue.

closed 2016-09-24 by Rotzbua