miguelbalboa/rfid #96: RC522 v2

indio99, 2015-04-06:
Hi

I have three models of RC522, I use your library and work fine, two chips have 0x80 and 0x91 versions, and work properly, but I have others with 0x92 version, and this only work with a 5% of all tag that I have.

Why??? In the datasheet say that 0x91 is version 1 and 0x92 is version 2 and the version 2 are better than version 1, but I can't do to work.

Thanks, best regards

omersiar, 2015-05-18:
Hardware may faulty.

norbertoramos, 2016-01-15:
Hi,
I have a similar problem.
Version 0x91 works ok.
Version 0x92 doesn't work with MFRC522::PICC_TYPE_TNP3XXX cards. The card isn't present. Hardware is ok.

Thanks and best regards.

norbertoramos, 2016-01-16:
I've supply 4.5V to the card and now it works.

norbertoramos, 2016-01-19:
as sugested in https://github.com/miguelbalboa/rfid/issues/168

Netoperz, 2016-01-24:
Yes, 4,5 and even 5V directly from Vcc pin of arduino should work,
There is a problem with range of that reader. NXP suggest using additional amplifiers. 
I just suggested using a higher voltage, for now, the module works constantly over a week without technical problems, but second prototype in completely closed case is getting a little worm, but nothing critical.

Right now i have got 4 prototypes build, and powered with 5v will be installed in a restaurant , as POS access control units for staff in place of old magstripe card readers. A friend agreed to give his restaurant  as testing field :) we will see if they work without problems.

Solution works on all versions of RC522

mdxs, 2016-05-16:
Can be closed.

miguelbalboa, 2016-05-16:
Thank you @mdxs

closed 2016-05-16 by miguelbalboa