miguelbalboa/rfid #98: Different types of rfid tags or protokols for them?

Skodra, 2015-04-11:
I want to scan my collage RFID tag, but i am not able. When i put my tag over the RC522 module nothing happens. Modul reds other tags with no problem.

So I am wondering are there different types of rfid tags or protocols for them and can i modify my modul or code in arduino to make it read this tag?

Rotzbua, 2015-04-11:
There are many possible differences.

See Wikipedia:
https://en.wikipedia.org/wiki/Radio-frequency_identification#Regulation_and_standardization

Compare to MFRC522:
https://github.com/miguelbalboa/rfid#protocols

Skodra, 2015-04-11:
Is there possibility in this library to change some mods, to se if different mods will work?

Rotzbua, 2015-04-12:
No. The MFRC522 is very cheap because the features are limited. Look at the data-sheet "2. General description" and you can read the physical functionality:
http://www.nxp.com/documents/data_sheet/MFRC522.pdf

closed 2015-06-16 by miguelbalboa