miguelbalboa/rfid #101: WARNING: Communication failure, is the MFRC522 properly connected?

brettbyrd, 2015-05-05:
I am new to ARDUINO. I have tried for the past two days to get the MFRC-522 to work with my Arduino UNO. I keep getting this message when I use the Dump Code 

********************
MFRC522 Software Version: 0x0 (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
********************
I set everything up just like in the sketch. 
Any help is appreciated

Rotzbua, 2015-05-07:
Does one of the other codes work? Do you use 3.3V? Try the RST to RST of the arduino.

danielyc, 2015-06-23:
i have the same problem, and tried those 3 things.
i hope there is a solution to this problem

mariohinojosa, 2015-06-24:
Hi, I am having exactly the same problem.

To give a bit of context. I have been using the RFID-RC522 for a project with a RPi, however when I changed it from a breadboard to a permaproto it stopped reading the cards. I wanted to check if the reader was the problem or the board I created, thus, I decided to test the reader on my Arduino Uno. I followed every step mentioned in the Readme file and in other YT videos but I am getting the aforementioned warning.

Could it be that the reader is very fragile and prone to breaking?

ccreutzig, 2015-07-12:
Did you ever link the reader directly to the Arduino? If so, you may have fried the inputs with your 5V levels. Not all 3.3V equipment tolerates getting 5V, even for a short time.

KiKiZ59, 2015-10-06:
Hi,
I've got the exact same problem. I verified my wiring 20 times, i connect the arduino to 3.3V.
I tried with UNO and MEGA and with two different RFID reader.
I always have this warning : 

MFRC522 Software Version: 0x0 (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?

I added these lines just below mfrc522.PCD_Init();	

        Serial.print("MFRC522 software version = ");
        Serial.println(mfrc522.PCD_ReadRegister(mfrc522.VersionReg),HEX);

And this returns 0.

Can anyone help me please? I'm getting mad with that problem.
I use an arduino clone bought in China that use ch340g driver.
Does the library work with that clone?
Thanks,
Kikiz

danielyc, 2015-10-06:
soldering the pins fixed the problem for me

closed 2015-12-05 by miguelbalboa

Tarino01, 2015-12-16:
I had exactly the same problem and the reason was that the connection of MOSI and MISO was reverse!
I used a pdf file from internet to determine the pinout of Arduino pro mini (3.3V, 8MHz). However, there was an error in that pdf file about MISO and MOSI pins. It is better to check your pins once again.

DoctorRu, 2015-12-19:
Remember to change code from default REST_PIN 9 and SS_PIN 10 to 5 and 53 if you are using in Mega 2560.

<code>
#define RST_PIN		5		</code>
<code>
#define SS_PIN		53	</code>

antararay19, 2016-01-13:
It could be a connection issue. In my case after fixing on the breadboard properly , when the red LED glowed consistently, the issue went away and version came

KiKiZ59, 2016-01-13:
I'd like to thank all of you for your responses.
I fixed the problem soldering the pins instead of using a breadboard.
kikiz

sarge8306, 2016-02-23:
Thank you for the suggestion to solder the pins....I finally got this thing to work now :).  I also wanted to thank the suggestion to increase the baud rate, I've tried tinkering with another project that had a slow response to no avail and was beginning to doubt my logic, but this reinforced it...thank you again!

TahasinAveins, 2017-11-25:
MFRC522 Software Version: 0x0 (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
 how can i config 
#define RST_PIN         9          // Configurable, see typical pin layout above
#define SS_PIN          10         // Configurable, see typical pin layout above

please help me

surabhi96, 2017-12-03:
@TahasinAveins check out the response by @DoctorRu in case you are using arduino mega. If you're using arduino uno you don't have to change the config.

sai-suresh, 2018-05-12:
soldering the pins will fix the problem.

brucesaints, 2018-07-16:
resold the pins fixed the problem for me too

sobinsunny111, 2018-10-29:
I used RFID-RC522 with Arduino UNO. I had the same issues but could not find any solution despite searching number of forums, videos and sites.

Finally, I resolved the issue by these steps:
    1. Make sure that you have installed the RFID library.
    2. If not installed, go to Sketch - Include library - Manage libraries - Search for RFID in the search box and install MFRC522 Library.
    3. If the RFID still dont work, then go back to the same step 2 and this time you might see an option for 'Update'.
    4. Do step 3 twice or thrice if required.
    5. Now go to Examples - MFRC522 - DumpInfo and upload it. Check the serial monitor. Your issue might get resolved as mine was okay by following this steps.

TahasinAveins, 2018-10-29:
Thanks for your reply..
But I have solved the problem.. It was loss connection

21700965, 2018-12-19:
How did you solve the problem without soldering the pins?

TahasinAveins, 2019-01-23:
No no, I have done my soldering..

YapWC, 2019-02-03:
Probably is the connection problem just change new wire or maybe just solder it up. This will solve the problem.

TahasinAveins, 2019-02-03:
Yep right

fab97419, 2019-06-10:
Hello everyone,
I'm sorry for the way I'm going to write, because I speak bad English I'm from France.
Regarding the communication error of the RIFD RC522 module, I had exactly the same problem and I just solved it by soldering the wires.
here I hope this will help you :)

mathiasconradt, 2019-08-08:
Confirmed, soldering the pins fixed it work me as well.

douglasdeodato, 2020-02-15:
2020 update  - Confirmed, soldering the pins fixed it for me as well.

I tried without solder and did not work, so I have to solder.

Kusiimwa, 2020-05-12:
> I used RFID-RC522 with Arduino UNO. I had the same issues but could not find any solution despite searching number of forums, videos and sites.
>
> Finally, I resolved the issue by these steps:
> 1. Make [...]

This has worked for me. Thank you!

Nisanth95, 2021-05-17:
I tried with jumper wires, but I still get the warning. 
Do I still have to buy new wires although my RFID-RC522 lights up when connected? 
Is there another solution than soldering? I need a breadboard to connect the arduino and RFID to a nodeMCU.
Thanks in advance

starfield33, 2021-05-17:
Reverse your communication cable - often when they’re around the wrong way they bring this error up. 

 
Very Best Regards, | 
Mit freundlichen Grüßen
Graham Andrews

***@***.***
***@***.***
‭‭+‭44 7787 133043‬ Mobile - London

Nisanth95, 2021-05-21:
I somehow made it work, but when I take my card og tag, it will not register. 
Although the warning line is gone, the firmware version is unknown. 
It says 
"?Firmware Version: 0xF8 = (unknown)
Scan PICC to see UID, SAK, type, and data blocks..."

SauravBhosale, 2021-11-05:
Guys Check your wires. I had the same problem. Try using new Jumper wires, it solved the problem in my case

Nip-pa, 2022-03-07:
MISO   >>>   D6
MOSI   >>>   D7

#define RST_PIN        D3        
#define SS_PIN          D4

AazWar-Tech, 2023-01-20:
Lo que a mi me funciono , es que precisamente no estaba del todo conectado, es decir que  si coloco el lector en la protoboard o lo dejo conectado en los pines pero sin soldar en el lector marcara la advertencia, pero si hago presión para que los pines toquen bien el circuito del lector entonces se quita el warning y funciona correctamente.

vashist22, 2023-04-19:
check the baud rate,set it to 115200

kariuki51jane, 2023-06-05:
Am experiencing this problem with my RFID-RC522, any one to help me out

WARNING: Communication failure, is the MFRC522 properly connected?
Scan PICC to see UID, SAK, type, and data blocks...
��@&4��p0� = (unknown)

RaduMinecraft, 2023-07-21:
On my genuine arduino doesn't work but on arduino clone works perfectly, why?

kyle-yeh, 2023-10-24:
It may be that the voltage is too low due to poor contact. I pressed it to make the board more in contact with it.

FADELELLAHerrami, 2023-11-21:
For me, the issue lies in NOT using the breadboard and soldering the pins of the RFID-RC522. that's the solution

PBonvang, 2024-03-14:
Had the same problem.

**Specs:**
Board: Elegoo Uno R3 (Arduino Uno replica)
RFID: RFID-RC522

**Solution:**
- Use female to male wires
- Instead of the v3.3 pin use the v_in pin

TxMike0038, 2024-04-24:
> Had the same problem.
> 
> **Specs:** Board: Elegoo Uno R3 (Arduino Uno replica) RFID: RFID-RC522
> 
> **Solution:**
> 
> * Use female to male wires
> * Instead of the v3.3 pin use the v_in pin

Connecting to the V_in instesad of the 3.3v fixed the issue for me, thank you so much!

achrafbjn74, 2024-10-02:
if you have a problem of MFRC522 not connected , try soldering places between wires and RC522 , the soldering is well

dofan28, 2024-12-29:
https://www.youtube.com/watch?v=5ZirOCEJ9AA 

i'm suggestions recommend watch it, it working me