miguelbalboa/rfid #104: rc522 -- reads cards but not fobs

rch33, 2015-05-13:
I've tried 2 522 devices 2 cards that came as samples. These worked. But the sample fobs did not read. I also had 5 other fobs (seemingly same kind) -- didn't read. Using miguelbalboa/rfid software. 

Also, not working: can't read a vendor card nor the chip in a valuable dog's ear.

Any suggestions?

Rotzbua, 2015-05-14:
> The MFRC522 reader supports only ISO/IEC 14443 A/MIFARE mode at 13.56 MHz electromagnetic field.

No other types or cips are supported. RFID is a general term for many frequencies and standards.

> mfrc522.PCD_SetAntennaGain(mfrc522.RxGain_max); //Set Antenna Gain to Max; will increase reading distance

Sometimes this helps.

rch33, 2015-05-14:
I tried your "SetAntennaGain" statement. It neither helped nor hurt. If the rc522s will only read the cards why would the vendors also send along sample key fobs?

Rotzbua, 2015-05-15:
Maybe he sent wrong forbs? Maybe the forbs are broken? Maybe the antenna of the rc522 and the antenna of the forbs are too bad? I don't know.
I got cards to my rc522 but they didn't work. I bought some from aliexpress and about 10% of the forbs didn't work.
You see sometimes there are some issues, but it is a problem of the hardware and not the software.

awyork, 2015-05-25:
I had a problem with my FOBs.  I really wish I could remember how I fixed it.  The problem was in the software.  This is what I think it was:

```c
// Authenticate using key B
    Serial.println(F("Authenticating again using key A..."));
    status = mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, trailerBlock, &key, &(mfrc522.uid));
    if (status != MFRC522::STATUS_OK) {
        Serial.print(F("PCD_Authenticate() failed: "));
        Serial.println(mfrc522.GetStatusCodeName(status));
        return;
    }
```

Note that for some reason I changed it use KEY_A.  I really think that's what fixed it, but it was so long ago.

rch33, 2015-05-25:
Thanks for your reply. However, your code doesn't "fit" with mine. Could you share your whole sketch?

Netoperz, 2016-02-06:
As long as fobs/cards are ISO/IEC 14443 A they should, work.
Antenna range is poor in RC522, in that case PN532 is better. But you can check if powering the shield with 4,5 -5v is helping, second, is: you buy from china on aliexpress = you are getting what you are paying for. do not expect the shield for 1$ with pins, fob and card and shipping costs in that price to be 100% tested and working :), as i do  buy a lot from china, i always buy 25% more than i need, that 25% is almost always a faulty part of devices, wrongly assembled, poor quality. ..... And, check version of your module. I bet it is different than V1/V2, i bet it is 0x50 or 0x11 - those are RC522 rip offs made by not exactly NXP :) mostly day work fine, but with only 2 kinds of tags, and with limited range, that's why the module + fob + card +pins + international shipping costs 1-1,5$ 

(that was just for clarification for some people expecting miracles)

mdxs, 2016-05-15:
@rch33: with the above information, do you think it is a software/library issue or perhaps just hardware as suggested by others?  If hardware, this issue can be closed.

rch33, 2016-05-16:
Sorry. I don't know if this is just a hardware issue. I have 9 fobs. None work. My 4 cards all work. But as I said. Why would the kits include both cards & fobs if only the cards can possibly work. Working fobs would make these devices much more useful.

mdxs, 2016-05-16:
In that case, you'll need to provide much more information on the readers, your Arduino; the wiring and which sketch you've tried (DumpInfo.ino might be a good start...).

closed 2016-08-25 by Rotzbua