miguelbalboa/rfid #109: Compilation issues after STM32 fixes with ESP8266

lcltech, 2015-06-15:
The current version gives at lot compile errors/ warnings when using it with the ESP8266. The reveret-76 works fine.

Rotzbua, 2015-06-16:
Too less information.. Share compiler version and those those error/warnings with us.

anistor, 2015-06-16:
@lcltech Please post your errors otherwise we can do very little to help. But since you mentioned ESP8266 I have a hunch this may be caused by lack of Arduino specific types like ```int8_t``` on the ESP8266. I have used that type in this commit https://github.com/miguelbalboa/rfid/commit/68f0b68a5aae725e3ab5a00678b719ae3a9515ff to make it work on the STM32 but did not anticipate this code is also used on non-Arduino platforms. Please try ```signed char``` instead of ```int8_t``` and let us know.

anistor, 2015-06-16:
@lcltech I've made a PR that replaces ```int8_t``` with ```signed char``` here https://github.com/miguelbalboa/rfid/pull/110 .  Please give a try to my branch. Thanks!

lcltech, 2015-06-16:
The Arduino IDE is a modified version 1.6.1. due to some reset issues with ESP8266.
I think the types may be the way to go.
The amount of compile errors is massive, here is a random section:

C:\Program Files (x86)\Arduino\libraries\rfid-5c67ee3ac33c953750ef8a4f26f6a3a37dd779e8\MFRC522.cpp:1721:44: error: call of overloaded 'println(char* const*)' is ambiguous
    Serial.println(GetStatusCodeName(status));
                                            ^
C:\Program Files (x86)\Arduino\libraries\rfid-5c67ee3ac33c953750ef8a4f26f6a3a37dd779e8\MFRC522.cpp:1721:44: note: candidates are:
In file included from C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Stream.h:26:0,
                 from C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/HardwareSerial.h:32,
                 from C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Arduino.h:212,
                 from C:\Program Files (x86)\Arduino\libraries\rfid-5c67ee3ac33c953750ef8a4f26f6a3a37dd779e8\MFRC522.cpp:7:
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:79:16: note: size_t Print::println(char) <near match>
         size_t println(char);
                ^
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:79:16: note:   no known conversion for argument 1 from 'char* const*' to 'char'
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:80:16: note: size_t Print::println(unsigned char, int) <near match>
         size_t println(unsigned char, int = DEC);
                ^
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:80:16: note:   no known conversion for argument 1 from 'char* const*' to 'unsigned char'
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:81:16: note: size_t Print::println(int, int) <near match>
         size_t println(int, int = DEC);
                ^
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:81:16: note:   no known conversion for argument 1 from 'char* const*' to 'int'
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:82:16: note: size_t Print::println(unsigned int, int) <near match>
         size_t println(unsigned int, int = DEC);
                ^
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:82:16: note:   no known conversion for argument 1 from 'char* const*' to 'unsigned int'
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:83:16: note: size_t Print::println(long int, int) <near match>
         size_t println(long, int = DEC);
                ^
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:83:16: note:   no known conversion for argument 1 from 'char* const*' to 'long int'
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:84:16: note: size_t Print::println(long unsigned int, int) <near match>
         size_t println(unsigned long, int = DEC);
                ^
C:\Users\LCL\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.h:84:16: note:   no known conversion for argument 1 from 'char* const*' to 'long unsigned int'
Error compiling.

anistor, 2015-06-16:
So all those compile errors happen with master but everything compiles ok with this older branch https://github.com/miguelbalboa/rfid/tree/revert-76-master ?

lcltech, 2015-06-16:
Yes, revert-76 compiles.
 I forgot to mention that I tried your modified version(int8_t  -> un. signed), but it did not seem to change any thing.

anistor, 2015-06-16:
Ok, I'll remove my branch. Thanks !

anistor, 2015-06-16:
@lcltech The compilation issues do not seem to be related to my STM32 fixes, and given that it works ok on revert-76 branch that makes me suspect moving strings to flash may have something to do with it (maybe @Rotzbua could help here). I believe the title of the issue should be updated to something like 'Compilation issues with ESP8266'.

Rotzbua, 2015-06-16:
Yes, it seems to bee a problem with the flash. The readme of esp8266 ide version says that there is a support of F("") macro (https://github.com/esp8266/Arduino#progmem).
I think the implementation of print() dosn't support const __FlashStringHelper* properly.

Rotzbua, 2015-06-16:
Sorry, I can't reproduce your errors. I tried several examples.
But I didn't used the old releases from https://github.com/esp8266/Arduino .
I uses the new 1.6.4 version from arduino website and installed esp8266 libs with the board manager: https://github.com/esp8266/Arduino#installing-with-boards-manager .

lcltech, 2015-06-16:
Removing the FlashStringHelper did the trick, now it compiles.
@anistor, I apologize and have changed the title.
@Rotzbua, it will see if I can try the 1.6.4 with out messing up my 1.6.1
The reason I am staying with the 1.6.1 is the auto reset issues with the serial monitor and some ESP8266 boards. Also I am not sure the boards manager def. files works with the Nodemcu v1.0 board that is connected to the RC522 board.

anistor, 2015-06-16:
Cheers @lcltech! No harm done, you don't need to apologize.

lcltech, 2015-06-16:
Well, now it gets a bit strange. I have installed the new 1.6.5 and the rfid master lib. compile error, damn!
Now i tried changing the board til generic esp8266, this board definition is from the same place as @Rotzbua gets it, now it compiles.
So it seem to boil down to a bad board def. This lib just was the first to use F()/Flashstringhelper.
Thanks for your help!

I will take the issue up here instead:
https://github.com/esp8266/Arduino/issues/387

closed 2015-06-16 by lcltech