miguelbalboa/rfid #111: RF 522 not working with Arduino Yun

programmin1, 2015-06-25:
The Arduino Yun has wiring a bit different than others, and though I followed the instructions here for wiring it to it, it isn't working:
http://arduino.stackexchange.com/questions/6697/arduino-yun-rfid-rc522-module-not-working

It compiles fine, but doesn't even pick up the RFID tag that came with the kit.

I assume there are some constants that I need to change somewhere to tell it that the pins are 5 pins on ICSP connector?

cryptid11, 2015-12-04:
Same problem here, did you solve it @programmin1 ?

programmin1, 2015-12-05:
Nope. I thought of trying it with a Raspberry Pi.

cryptid11, 2015-12-05:
So, really, no solution to use it with yun, but the connections are correct for sure! This image confirms it http://www.14core.com/wp-content/uploads/2015/06/Arduino-Yun-PinOut.png that's really too stupid to be true, please somebody with some expertise solve this issue!

Rotzbua, 2017-01-16:
@programmin1 @cryptid11 Can I close this? Did you found a solution?

cryptid11, 2017-01-16:
1 - You decide
2 - No

Rotzbua, 2017-01-16:
@cryptid11 thanks for feedback. I added it to the readme. Maybe someone will fix this in future...

closed 2017-01-16 by Rotzbua

cmddegi, 2018-09-11:
Hi! I just hooked up a Joy-It RFID-RC522 with my *Yun* and it *worked instantly* without changes to the code. You just have to get the wiring right. On the Yun (and on the leonardo) the *SPI is available only on the ICSP header*:

This works for me:
* MISO on ICSP1
* VCC on ICSP2
* SCK on ICSP3
* MOSI on ICSP4
* (ICSP5 is free)
* GND on ICSP6
* RST on Pin9 (default)
* NSS on Pin10 (default)

Thanks for the library and the examples! Worked in no time. :-)

Rotzbua, 2018-09-11:
@cmddegi Thanks for your positive feedback :+1:

PaulBeaudet, 2020-06-03:
I've got this working on the Yun in the past as well, going to remove this from the readme

Rotzbua, 2020-06-03:
@PaulBeaudet Was your wiring like https://github.com/miguelbalboa/rfid/issues/111#issuecomment-420433658 or did you use a different wiring?

PaulBeaudet, 2020-06-03:
@Rotzbua Thanks for maintaining this library btw. Having set up our system with it I'm grateful it continues to be supported

unfortunately @programmin1 @cryptid11 may have got bad boards. Did either of you test with other hardware?

dimact22, 2021-10-04:
tell me the pins for connecting the Esp32 Devkit V1 board

commit 0b91afb4 (2017-01-16): Update README.rst

close #111
close #240