miguelbalboa/rfid #113: Sketch for detecting presence of the card - could you please explain?

CRImier, 2015-07-12:
Hello!

I bought some RC-522 readers for a device that needs to sense when an RFID card with a specific UID is present on it (turning off as soon as card is removed), so the debug version of sketch would be constantly checking whether card is there or not and sending this data to Serial. I've started with a DumpInfo example, which worked flawlessly, then read through .h and .cpp library files. I've come up with this sketch which works as intended:

```
boolean card_present = false;

void loop() {
  mfrc522.PICC_ReadCardSerial(); //Always fails
  mfrc522.PICC_IsNewCardPresent(); //Does RequestA 
  if (! mfrc522.PICC_ReadCardSerial()) { //Okay. This does the same PICC_Select as the previous ReadCardSerial(), but this one fails if there is no card on the reader. Funny. 
    //Seems like we need two Select's in a row to detect card's presence.
    card_present = false;
  }
	else {
    card_present = true; //variable for my own purposes
    print_uid(&(mfrc522.uid)); //prints UID in human-readable form for debugging
    uid = mfrc522.uid;
	}
  print_presence(); //prints "Card present" or "Card not present"
  delay(1000);
}
```
I don't really understand why should there be two Select calls, but without the first one only every second iteration of loop() produces a valid result. The first call seems to always fail (not STATUS_OK). I understand  that the first Select must be calling some lower-level function that makes it fail but resets something so it detects the card the next time Select() is called. What would that function be? Or is it that I'm just using it wrong? 

Also, once that's clear - would that be any good for an example sketch? I'd gladly make a pull request =)

Rademanc, 2016-02-05:
Hi.
Yeah I have the same problem and used the same approach, though it seemed to not always detect the card if I call those methods twice. Any luck?

CRImier, 2016-02-05:
Hi, @Rademanc!

In the end, I've made a RFID reader script working as an I2C slave. Worked perfectly for me in the 'production environment', and you could insert some serial debug statements for you to check whether this works for you the same way it does. 

```
#include <SPI.h>
#include <Wire.h>
#include <MFRC522.h>
#define RST_PIN		A1		
#define SS_PIN		A0		

#define address 2

uint16_t xor_uid = 0;

MFRC522 mfrc522(SS_PIN, RST_PIN);	// Create MFRC522 instance

void setup()
{
	SPI.begin();			// Init SPI bus
	mfrc522.PCD_Init();		// Init MFRC522
	Wire.begin(address);                // join i2c bus with address #8
  Wire.onRequest(sendCardInfo); // register event
  //mfrc522.PCD_SetAntennaGain(mfrc522.RxGain_max);
}

uint16_t read_card() {
  uint16_t xor_uid = 0;
  if (! mfrc522.PICC_ReadCardSerial()) 
    { //Okay. This does the same PICC_Select as the previous ReadCardSerial(), but this one fails if there is no card on the reader. Funny. 
      //Seems like we need two Select's in a row to detect card's presence.
      xor_uid = 0;
    }
  else 
  {    
    for (int i = 0; i < mfrc522.uid.size; i=i+2) 
    {  
      xor_uid = xor_uid ^ (mfrc522.uid.uidByte[i]<<8 | mfrc522.uid.uidByte[i+1]);
    }
  }
  return xor_uid;
}

void loop()
{
  mfrc522.PICC_ReadCardSerial(); //Always fails
  mfrc522.PICC_IsNewCardPresent(); //Does RequestA 
  xor_uid = read_card();
  delay(1);
}

void sendCardInfo() {
  byte myArray[2]; //Thanks http://thewanderingengineer.com/2015/05/06/sending-16-bit-and-32-bit-numbers-with-arduino-i2c for this solution
  myArray[0] = (xor_uid >> 8) & 0xFF;
  myArray[1] = xor_uid & 0xFF;
  Wire.write(myArray, 2);
}
```

Rademanc, 2016-02-06:
Thanks a lot.
I will give it a shot.

mdxs, 2016-05-16:
Can be closed.

closed 2016-05-16 by miguelbalboa

zhekaus, 2019-11-27:
Actually, this feature still exists. We need to call PICC_ReadCardSerial extra time to get correct result.