miguelbalboa/rfid #117: Multiple tries to authenticate

davyvr, 2015-08-05:
Hi there,

I have a bit of a problem with authentication. I want my program to find out if a card has the default authentication key as KeyB (called KeyDefault in my software) or a key I made myself (KeyPrivate). So I want to try both authentications, and if default rewrite the authentication. For this I made this function:

Serial.println("Logging in to RFID tag ...");
  byte TagType = mfrc522.PICC_GetType(mfrc522.uid.sak);
  if (TagType != MFRC522::PICC_TYPE_MIFARE_1K)
  {
    Serial.println("Wrong tag type! Use MiFare 1k tags");
    Serial.println("");
    return false;
  }

  byte status = mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, trailerBlock, &KeyDefault, &(mfrc522.uid));
  if (status!= MFRC522::STATUS_OK)
  {
    Serial.println("Tag security key A corrupt, throw away!");
    Serial.println("");
    return false;
  }
  
  status = mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_B, trailerBlock, &KeyDefault, &(mfrc522.uid));
  if (status!= MFRC522::STATUS_OK)
  {
    mfrc522.PCD_StopCrypto1();
    status = mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_B, trailerBlock, &KeyPrivate, &(mfrc522.uid));
    if (status!= MFRC522::STATUS_OK)
    {
      mfrc522.PCD_StopCrypto1();
      Serial.println("Tag security key B corrupt, throw away!");
      Serial.println("");
      return false;
    }
    else
    {
      SetAccess = false;
      return true;
    }
  }
  else
  {
    SetAccess = true;
    return true;
  }

The problem is: PICC_CMD_MF_AUTH_KEY_B doesn't work if KEYB is equal to KeyPrivate. If I switch both places (KeyDefault & KeyPrivate) then it works. So the first time I try to authenticate KeyB works, but the second time doesn't. How can I try two different keys for KeyB? Am I making a syntax error or forgetting something?

Kind regards,
Davy Van Rossem

norbertoramos, 2016-01-16:
Hi
I solve the problem calling
            // Reinicia la tarjeta
            mfrc522.PICC_IsNewCardPresent();
            mfrc522.PICC_ReadCardSerial();          
After each try.
Regards

mdxs, 2016-02-28:
@davyvr Is this still a problem for you (after trying the solution proposed by @norbertoramos)?  Otherwise, I think this ticket should be closed.

mdxs, 2016-05-15:
Can be closed, no response must mean all is fine.

closed 2016-05-15 by miguelbalboa