miguelbalboa/rfid #122: Mifare DESFire

yoprogramo, 2015-08-26:
Hello, I know that the card can read Mifare DESFire tokens but the library not... Do you plan to include it or can I help on implementing it?

miguelbalboa, 2015-08-26:
Hello Jose, 
The true is that right now i'm not working in RFID technology anymore, any Pull request will be welcome, so, be my guest.

Thank you.

yoprogramo, 2015-08-26:
Hello Miguel,

I'm an experienced programmer, BUT the problem is I have no idea where to start to complete the library. Please, give me some gidance and I'll try to program the rest... (even in Spahish if it make things easier). I only work in RFID eventually for a project but I'm very much interested (and I have the hardware), so I'll try.

mlc73, 2016-02-23:
Really nobody's tried deploy DesFire?

yoprogramo, 2016-02-26:
In my case I have not worked before with NFC technology, I only integrate
the results... So, I have no idea on how to start...

avanderpluijm, 2016-11-18:
Hi, 
This code seems to work: http://www.codeproject.com/Articles/1096861/DIY-electronic-RFID-Door-Lock-with-Battery-Backup 
Could that be something to integrate? It's a bit above my current knowledge level to do it myself I am afraid... 

and https://github.com/nceruchalu/easypay,

vbonamy, 2016-11-19:
@avanderpluijm :  http://www.codeproject.com/Articles/1096861/DIY-electronic-RFID-Door-Lock-with-Battery-Backup  is interesting but that's about PN532 and not MFRC522.

avanderpluijm, 2016-11-19:
ok I understand, thank you, the difference between PN533 and MFRC522 was not clear to me. 
Still, couldn't you use the Desfire logic from that code to use in this library? 

I am eager to do this, but any help is appreciated!

Rotzbua, 2016-11-19:
The main problem is, that NXP do not allow to publish the specification of crytpo of any Mifare product.

The library you mention is based on reverse engineering and some documents which are found (maybe illegal) on the internet. This means if you try to integrate the code in this project, it would be really difficult to debug, because there are official information from NXP available.

The other problem is ram and flash consumption. AES takes a lot, so I think the code would not work on the old Atmel Cips, but on Arduino Mega, M0, teensy ... But we could solve this by compiler flags.

avanderpluijm, 2016-11-21:
So it needs to be rewritten to MFRC522; 
Yes, in my case I am using an ESP8266, a classic arduino is too small. 

Is the reason that you are not allowed to publish specs (when you sign an NDA) a reason not to accept a pull request for this repository?

Rotzbua, 2016-11-21:
> Is the reason that you are not allowed to publish specs (when you sign an NDA) a reason not to accept a pull request for this repository?

No. I did not signed any NDA, so I do not care about a NDA.

JPG-Consulting, 2017-01-10:
Hi,

Talking about NDA I'm not sure if MIFARE DESFire code is welcomed in the library. Is it welcome?

@avanderpluijm I've only submitted a pull request (https://github.com/miguelbalboa/rfid/pull/271) adding some ISO/IEC 14443 methods (RATS and PPS). Both this methods are used while identifying a MIFARE DESFire card. Since I'm not sure MIFARE DESFire code is welcomed I've removed all DESFire code from that pull request. Right now I've been able to send a `GetVersion` command identifying the card (Version number for both hardware and software, storage size,...)

Right now I was thinking about creating a new class whichinherited MFRC522 to add this features... but if it is welcomed here, maybe I could share some of that code. I'm not too sure how an Arduino UNO can handle DES/AES (Haven't reached that point yet) but the initial stages can certainlly be worked out with a MFRC522 and Arduino UNO (That is my test rig).

shyaminayesh, 2018-01-10:
@JPG-Consulting any updates on your work ? seems lib is not yet support for read DESFire :(

PeterTonoli, 2019-04-11:
Ping @JPG-Consulting