miguelbalboa/rfid #124: MIFARE ULTRALIGHT CARD NOT DETECTED

eengrhassaan, 2015-09-04:
Hello, 
I musing MFRC522 with Arduino Mega, I m able to detect MIFARE classic cards and tags but unable to detect the MIFARE ULTRALIGHT CARD and TAGs. Will you please guide me on this. ?????

mrStanislav, 2015-10-14:
Good evening.
I also had this problem. I found one way, instead 3.3volts I connected to 5 volts. its work for me. you can check also. As I remember, the module is tolerant to 5 volts. anyway my module works and reads all cards like Mifare: 1k, mini, ultralight etc..

eengrhassaan, 2015-10-16:
but it doesn't support ULTRALIGHT C 3DES Authentication

Rotzbua, 2015-10-16:
Only unencrypted communication is supported by this library.
DES is not supported by hardware, it has to be implemented in software but nobody done yet.
It would be nice if you implement and make a pull request :)

closed 2015-12-05 by miguelbalboa