miguelbalboa/rfid #126: Two boards bought at amazon do not work - am I missing something?

OliverTED, 2015-09-04:
Hello Miguel (and others),

I bought two boards:
http://www.amazon.de/gp/product/B00J2SS2U6
http://www.amazon.de/gp/product/B00QFDRPZY
and both do not work.

Connection appears to be working as they report versions (0x92 and 0x88). Now I am running the program:
rfid/examples/DumpInfo/DumpInfo.ino

The call to 
mfrc522.PICC_IsNewCardPresent()
fails in MFRC522.cpp:418 after iterating for 2000 times (on both cards).

I tried increasing the counter up to 2000 * 1000 / 35 (1 second?). And iterations are still exhausted.

I am new to electrics and curious what I do wrong.  Could I break them soldering ?  Is the quality so bad both could be broken on arrival?  I would be thankful for any hints.

chrisstaite, 2015-09-11:
Sounds like you are having the same trouble I was having.  Take a look at #127

Bladeslayer1000, 2015-09-11:
There could be a bat solder joint on the main IC on the RC522 you can fix it, but its quite difficult. 
You can try to resolder the header pins, make sure it looks like its connected.

A other possibility is that you're connecting the module in the wrong way. Make sure all the pins match use the RFID-RC522 - Pin Layout, you can find it in; 'rfid-master -> doc -> fritzing'

OliverTED, 2015-09-19:
Chrisstaite:  I did try your code, but it did not help.  Actually self tests always succeed with both boards. Repeated initialization did not change anything.

Bladeslayer:  I wonder, if I connect them in the wrong way, is it possible that self tests succeed and versions are reported?  Reported versions are (0x92 and 0x88) and differ between my two boards.

MarcoBlulab, 2015-10-02:
I've bought similar item on Amazon Italy. The card reader was not working at all. 
After countless tests I realized that the problem was in the solder. I redid all the soldering of the RC522  chip (with a normal soldering iron, it takes a lot of patience and a steady hand...) and now works like a charm.

closed 2015-12-05 by miguelbalboa