miguelbalboa/rfid #127: FIXED: Board not passing test

chrisstaite, 2015-09-11:
This isn't an issue, but an issue I had and fixed.  Without anywhere better to record it, I thought I'd add it here so someone else with the same problem might be able to fix theirs too.

After testing, the board was working great.  I bought a few more for scaling up the project, however the new boards were failing the self test saying DEFECT or UNKNOWN although it was reading the version correctly (v2.0).

When I investigated it read reading three NULL (0) bytes before reading the correct reference data for the chip during the selftest.  It also did not return the last 5 reference bytes, instead finishing with two NULL bytes.

After a while I started running the tests in a loop to figure out what was happening.  After it failed the first time, it succeeded the second, but still couldn't read cards.  Then I decided to initialise the card reader again after a successful test.

The solution is as follows:

  mfrc522.PCD_Init();
  
  bool hasFailed = false;

  while (!mfrc522.PCD_PerformSelfTest())
  {
    hasFailed = true;
    delay(100);
  }

  if (hasFailed) {
    mfrc522.PCD_Init();
  }

Hope this helps.

Bladeslayer1000, 2015-09-11:
I have changed it a bit, so its more obvious if the RC522 module has failed or not.

The code:
```/*
 * Self test
 * Uses MFRC522 - Library to use ARDUINO RFID MODULE KIT 13.56 MHZ WITH TAGS SPI W AND R BY COOQROBOT. 
 * This test is created by BladeSlayer1000 inspired by the code of chrisstaite https://github.com/chrisstaite
 * -----------------------------------------------------------------------------------------
 *             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
 *             Reader/PCD   Uno           Mega      Nano v3    Leonardo/Micro   Pro Micro
 * Signal      Pin          Pin           Pin       Pin        Pin              Pin
 * -----------------------------------------------------------------------------------------
 * RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
 * SPI SS      SDA(SS)      10            53        D10        10               10
 * SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
 * SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
 * SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15
 *
 * Hardware required:
 * Arduino
 * PCD (Proximity Coupling Device): NXP MFRC522 Contactless Reader IC
 * PICC (Proximity Integrated Circuit Card): A card or tag using the ISO 14443A interface, eg Mifare or NTAG203.
 * The reader can be found on eBay for around 5 dollars. Search for "mf-rc522" on ebay.com. 
 */

#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN 10    //Arduino Uno
#define RST_PIN 9
MFRC522 mfrc522(SS_PIN, RST_PIN);        // Create MFRC522 instance.
bool hasSaid = false;
bool hasFailed = false;

void setup() {
        Serial.begin(9600);        // Initialize serial communications with the PC
        SPI.begin();                // Init SPI bus
        mfrc522.PCD_Init();        // Init MFRC522 card
        Serial.println(F("Running RC522 self test"));
}

void loop() {
	while (!mfrc522.PCD_PerformSelfTest()) {
		hasFailed = true;
		delay(100);
	}

	if (hasFailed) {
		if(hasSaid == false) {
			Serial.println("Test failed, please check your RC522 module");
			mfrc522.PCD_Init();
			hasSaid = true;
		}
	} 	else {
		if(hasSaid == false) {
			Serial.println("Test passed");
			hasSaid = true;
		}
	}
}

closed 2015-12-05 by miguelbalboa

skyguy126, 2018-05-13:
@JRTax how did you fix this issue, the code you pasted here makes no output to serial monitor.

VladC12, 2020-03-15:
For some reason, I am stuck in the while loop.

marha915, 2020-07-10:
I'm also stuck in the while loop with my cheap blue modules (blue PCB marked  "RFID-RC522" and "HW-126", unmarked chip)
Works like a charm with the "fancier" model from Velleman. (marked "VMA405" and with genuine looking chip)
Did anyone make any progress here?

garinuss, 2022-02-24:
> I'm also stuck in the while loop with my cheap blue modules (blue PCB marked "RFID-RC522" and "HW-126", unmarked chip) Works like a charm with the "fancier" model from Velleman. (marked "VMA405" and with genuine looking chip) Did anyone make any progress here?
> 
> Edit; This is the result read from the FIFODataReg for me, doesn't seem to match any other reference I can find:
> 
> ```
> 0x00, 0x2E, 0xC6, 0xB5, 0x56, 0xA5, 0xF1, 0x71, 
> 0x91, 0xFD, 0xFD, 0xB1, 0xDE, 0x22, 0x7D, 0x07, 
> 0x7D, 0x48, 0xA5, 0x75, 0xBC, 0x39, 0x59, 0x53, 
> 0x6B, 0x69, 0xAF, 0xDA, 0xC3, 0xB8, 0x86, 0x3C, 
> 0xA0, 0xEF, 0x1E, 0xD3, 0xCC, 0xE7, 0xB2, 0xA4, 
> 0xEA, 0xFC, 0x51, 0x3E, 0x77, 0x9A, 0x71, 0x6B, 
> 0x92, 0x9D, 0x88, 0x6E, 0x82, 0xC9, 0x7A, 0xCF, 
> 0x3A, 0xEC, 0xD7, 0xD2, 0xC4, 0xF6, 0xDC, 0x0C,
> ```

I'm having the same behavoiur from a device with no markings. Searching i find this chip: VRC522 producer VIBRATION that seems to have no markings, but i can't find the datasheet. 

Did someone make any progress on this chips?