miguelbalboa/rfid #130: Can use RC522 after initializing SDcard.

manowarkdt, 2015-09-29:
I'm trying to use an SD Card and RC522 on same board, separatelly both works ok, I have both dedevices on different CS pins, and even ensure to HIGH the other CS pin before trying to use the RC522 but always fails.

SPI config seems to be the same for both,   MSBFIRST and SPI_MODE0, RC522 doesn't have option to change speed but initializing sd card at full speed, half speed (default for SPI library) or even quarter speed, have the same result.
I've tried to use mfrc522.setSPIConfig(); after using sd card to reconfig spi bus and neither. 

Any idea?

Regards

simbadMarino, 2015-09-30:
Hello manowarkdt, I'm actually using both modules as you want to, It is hard to know whats going on without your source code. Take a deep look in my project if you want.... maybe you get some ideas from my code :) https://github.com/simbadMarino/Odologger/tree/master/Odologger.

If you need further help do not hesitate to ask... Good luck!

manowarkdt, 2015-09-30:
Hi simbadMarino, no way,  I've tested all, once I initiate the SD card, MC522 doesn't work. I use Teensy 3.1 my CS pins are 14 for MC522 and 22 for SD Card.

Thank you

AxissXs, 2015-09-30:
Hi guys
i've exactly the same problem
i cut the sd reader power and rc522 works, i put it there and it stops working
i'm using uno
and i tried a hundred different ways, i don't think it's fixable by software :(

manowarkdt, 2015-09-30:
Hi, I got it working, in my case was a problem with SD library for my Teensy, after deleting from /hardware/teensy/avr/libraries the IDE use the default located in /libraries folder. 

I've notice that the problem was in that librarie because I have a third SPI device, a serial Flash memory and neither works with sd  card, so the problem was the SD card library not RC522, all works but after initializing SD card all fails.

All people claiming to work SD + RC522 was in Arduino not Teensy. So problem solved deleting teensy library.

AxissXs as i've read in few places with arduino even if you don't use pin 10 for SS you must set it as output. And if you use a shield ensure to set HIGH other SS pins used by this shield with SPI. 

In  my tests the my Flash chip id is  BF 25 4B, using faulty SD library the id reported was FF 25 4B and with rc522 the firmware id was 3F, no FF as when there is no communication at all. 

So if you don't have any data from rc522 maybe some config problem, other time I've set RC522 SS pin as input some where in the code because of code reused from other project where this pin had another use, this makes unstable, sometimes rc522 works some times not.

Regards, thanks to all and hope my experience helps.

closed 2015-12-05 by miguelbalboa

jonnygraham, 2016-04-13:
I also had this problem using RC522 and an SD card reader with an Arduino UNO. I worked out that if I disconnected the SD's MISO pin, the RC522 still worked.
So I tried to isolate them by adding a diode on each MISO line. The diode on the RC522's line stopped the RC522 from working at all (I'm guessing that this is because it's on 3.3V and the diode drops the signal by 0.7V and that's too low to be considered 'high' by the 5v Arduino) so I just left the other diode in and it solved the problem. I have to admit that I'm not sure why it helps. 

To summarize:
I got it to work by adding a diode between the SD card reader's MISO pin and the Arduino's MISO input (pin 12 on the UNO). The diode's direction is anode to SD Reader, cathode to Arduino.

Hope this helps!

shyamsingla11, 2016-07-22:
Hey guys I am shyam and I got the same problem but now I got the solution.
The logic less and the most simple solution for this is that give a seperate power supply of 3.3v for sd card , it will definitely work as well.

sumit5ue, 2017-07-24:
Hi Shyam - I am struggling with the same issue. Do you have a pic/schematic ? did you just provide a separate power supply or also added a diode?

dewinsmg, 2017-10-17:
Debes apagar y encender el pin 2 que va conectado al VCC del lector SD
[https://i.stack.imgur.com/dHVNO.jpg](url)
[https://i.stack.imgur.com/ltWt3.jpg](url)
`//HEADER FILES##################################
#include <SPI.h>
#include <MFRC522.h>
#include <SoftwareSerial.h>
#include <EEPROM.h>
#include <SD.h>
//ARDUINO MEGA
//MISO 50
//MOSI 51
//SCK  52
//ON/OFF SD CARD READER 2
//SD Card DEfinitions###########################
File root;
const int chipSelect = 4;
const int chipONOFF = 2;
//RF_ID Definitions#############################
#define SS_PIN 7   
#define RST_PIN 6
MFRC522 mfrc522(SS_PIN, RST_PIN);

void setup() 
{
    Serial.begin(9600);
    SPI.begin();  
    pinMode(chipONOFF,OUTPUT);   
    digitalWrite(chipONOFF, LOW);
    tc_03_rfIDbasicTest();
    digitalWrite(chipONOFF, HIGH);
    tc_02_SDcardConnectTest();
    digitalWrite(chipONOFF, LOW);
    tc_03_rfIDbasicTest();
}

void loop() 
{ }

void tc_02_SDcardConnectTest()
{
    Serial.println("");
    Serial.println("");
    Serial.println("-----TEST 2: FUNCIONAMIENTO BASICO DE SD-----");
    Serial.println("");  
    if (!SD.begin(chipSelect)) 
    {
        Serial.println("Falla en comunicacion, asegurate que la uSD esté insertada e intenta de nuevo");
        Serial.println("FAIL");    
    }
    else
    {
        Serial.println("Tarjeta uSD Detectada");
        Serial.println("PASS");
        root = SD.open("/");
        printDirectory(root, 0);    
    }  
}

int tc_03_rfIDbasicTest()
{   
    Serial.println("-----TEST 3: FUNCIONAMIENTO BASICO DE RFid-----"); 
    mfrc522.PCD_Init();                             // Init MFRC522 chip again per each write attemp  
    mfrc522.PCD_DumpVersionToSerial();              // Show details of PCD - MFRC522 Card Reader details

    if ( ! mfrc522.PICC_IsNewCardPresent()) 
        Serial.println("Targeta fuera de alcance");    
    else
        Serial.println("Targeta dentro de alcance");
    if ( ! mfrc522.PICC_ReadCardSerial()) 
        Serial.println("Falla en lectura de tarjeta");    
    else
    {
        Serial.println("RFid OK :) !");
        Serial.println("PASS");
        String nuid="";
        for (byte i = 0; i < 4; i++) 
            nuid+=mfrc522.uid.uidByte[i];
        mfrc522.PICC_HaltA();
        mfrc522.PCD_StopCrypto1();
        Serial.println(nuid);
    }       
}

void printDirectory(File dir, int numTabs) 
{
    while (true) 
    {
        File entry =  dir.openNextFile();
        if (! entry) 
            break;        
        for (uint8_t i = 0; i < numTabs; i++) 
        {
            Serial.print('\t');
        }
        Serial.print(entry.name());
        if (entry.isDirectory()) 
        {
            Serial.println("/");
            printDirectory(entry, numTabs + 1);
        } 
        else 
        {          
            Serial.print("\t\t");
            Serial.println(entry.size(), DEC);
        }
        entry.close();
    }
}`

esto es lo que se ve en el monitor serial

`-----TEST 3: FUNCIONAMIENTO BASICO DE RFid-----
Firmware Version: 0x92 = v2.0
Targeta dentro de alcance
RFid OK :) !
PASS
215103959

-----TEST 2: FUNCIONAMIENTO BASICO DE SD-----

Tarjeta uSD Detectada
PASS
BIO/
	D70A273B.DB		44
	ENROLADO.DB		57
	F683B22D.DB		40
	A6DE1E49.DB		38
	080732DC.DB		38
	08D1EF01.DB		38
	08267788.DB		38
XWIFI.DB		47
CP0101.DB		26
XFEC.DB		4
CP0102.DB		26
CP0201.DB		26
SYSTEM~1/
	WPSETT~1.DAT		12
	INDEXE~1		76
CP0301.DB		26
GP01.DB		33
GP02.DB		33
GP03.DB		33
GP04.DB		32
GP05.DB		32
GS06.DB		35
GS07.DB		37
GS08.DB		36
GS09.DB		36
GS10.DB		36
GE03.DB		24
GE02.DB		20
GE01.DB		23
GS11.DB		38
AP010101.DB		69
AP010102.DB		83

-----TEST 3: FUNCIONAMIENTO BASICO DE RFid-----
Firmware Version: 0x92 = v2.0
Targeta dentro de alcance
RFid OK :) !
PASS
215103959`

21jerks, 2018-09-22:
i am having a huge problem PLEASE HELP!!!
i set up my rc522 to the arduino uno, i downloaded the library that i needed. downloaded the rfid master, downloaded the arduino IDE. when i go to dump info, open up COM3 to dump the info, i try to scan ther IC CARD but no information dumps at all. it only tells me to scan the rfid to see UID, BLOCKS, AND OTHER INFO. but when i scan the chip to the rc522 nothing happens. i checked all the wires there hooked up to where they need to be, and still nothing. PLEASE HELP ME!!!

TiAmoPlus, 2019-05-06:
Is there a similar code to successfully make RFID and SD work together? I make them work by controlling the VCC of SD or RFID. Thouth it worked.

shyamsingla11, 2019-05-06:
If you are struggling with the problem that sd card module and rc522 rfid reader are not working together but work seperately,,
The solution for this is that use different power supply to power your sd card module and make the ground common...
This is caused due to high power consumption of sd card module and rfid module which drops the voltage and circuit is unable to detect any one of sd card module and rfid reader.. :)

TiAmoPlus, 2019-05-07:
> If you are struggling with the problem that sd card module and rc522 rfid reader are not working together but work seperately,,
> The solution for this is that use different power supply to power your [...]
Thanks for your help. I would try the method. 
And I maybe find another possible reason.
Here is official part of the sample _CardInfo_ from SD
`if (!card.init(SPI_HALF_SPEED, chipSelect)) {
    Serial.println("initialization failed. Things to check:");
    Serial.println("* is a card inserted?");
    Serial.println("* is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");
    while (1);
  } else {
    Serial.println("Wiring is correct and a card is present.");
  }`
It sets the speed of SPI to SPI_HALF_SPEED. 
`/** Set SCK rate to F_CPU/4. See Sd2Card::setSckRate(). */
uint8_t const SPI_HALF_SPEED = 1;`
Are they the same?
SPISettings can set some settings of SPI. Should I change the settings if not?
I am not good at English.