miguelbalboa/rfid #131: Software Version Unknown

KiKiZ59, 2015-10-06:
Hi,
I've got a probleon. A lot of people had the same but i couldn't find any solution on internet.
The problem is that the software version is unknown so there's no communication. 
I verified my wiring 20 times, i connect the arduino to 3.3V.
I tried with UNO and MEGA and with two different RFID reader.
I always have this warning (DumpInfo) :

MFRC522 Software Version: 0x0 (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?

I added these lines just below mfrc522.PCD_Init();

    Serial.print("MFRC522 software version = ");
    Serial.println(mfrc522.PCD_ReadRegister(mfrc522.VersionReg),HEX);
And this returns 0.

Can anyone help me please? I'm getting mad with that problem.
I use an arduino clone bought in China that use ch340g driver.
Does the library work with that clone?
Thanks,
Kikiz

danielyc, 2015-10-06:
have you soldered the connector pins to the scanner? because that was my problem

KiKiZ59, 2015-10-06:
Hi,
Thanks for your quick response. I soldered pins of the RFID reader but i use a solderless breadboard to make the communication between reader and arduino. Do you think using a solderless breadboard could cause the problem?
[image]
[image]

simbadMarino, 2015-10-06:
Try soldering the most you can, i also had a lot of issues because of big wires and solderless boards.  Hope it Works!
--
老的不去，新的不来… martes,

KiKiZ59, 2015-10-06:
Okay, I'm out of home for a few days so i'm not able to solder right now.
I'll try this in 6 days then i'll post a comment to tell you if that worked.
Thanks

KiKiZ59, 2015-10-13:
Hello guys,
I tried to solder the RFID reader instead of using solderless breadboard and now it's working.
DumpInfo recognize the version of the reader.
Thank YOU

danielyc, 2015-10-13:
Glad we could help!

closed 2015-12-05 by miguelbalboa