miguelbalboa/rfid #132: MiFare Ultralight Tag

KiKiZ59, 2015-10-14:
Hello,

I'd like to read and write on ultralight tag.
I only found sketch on internet to do it on classic cards.
I know that won't work because the UID of ultralight and classic
have different sizes.
Does anyone konw how to change the code on the examples of the library
to adapt it to ultralight?

Thanks,
Kikiz59

Simgy, 2016-06-18:
Hello, have you found a solution ? Because I am facing the same problem. I would like to read a ultralight tag but this one has no key.

JPG-Consulting, 2017-01-20:
Sorry I don't have an Ultralight card :(

While reading about DESFire there was some info about Ultralight stating some of the methods should be compatible for both Ultralight and DESFire cards. I was focusing on DESFire and didn't pay much attention to Ultralight commands or the compatibility between the commands I've written. However I think the basics was more or less the same:

- Send Request for ATS (RATS)
- Send PPS.

...and you should now be in T=CL. 

The ATS response tells you if you should use CID and/or NAD in the prologue of the block. So at this point you should be building the blocks to send MIFARE commands to the Ultralight. It's here where things changes as I think Ultralight uses pages and not ApplicationID (AID) nor files... So probably the next step with an Ultralight will be authentication... and if I'm not misteaken you'll need DES encryption there... I think the authentication command where compatible between Ultralight and DESFire (Although in ultralight it would use DES/3DES, not having the additional AES as the DESFire).

leandre84, 2017-08-25:
As the memory structure and commands of Mifare Ultralight are very similar (or rather the same) to NXP NTAG 21x, the following code I use for reading blocks from NTAG 21x may come handy:

```
/* Read 16 bytes from NTAG, starting with blockAddr */
int NTAGReadBlock(byte blockAddr, uint8_t *data) {
	byte length = NTAG_BLOCKLENGTH+2;
	byte buffer[NTAG_BLOCKLENGTH+2] = { 0 }; // this holds CRC as well

	MFRC522::StatusCode status;

	if ((status = mfrc522.MIFARE_Read(blockAddr, buffer, &length)) != mfrc522.STATUS_OK) {
		return -1;
	}
	if (length != NTAG_BLOCKLENGTH+2) {
		return -2;
	}

	memcpy(data, buffer, NTAG_BLOCKLENGTH);

	return 0;

}
```
```
#define NTAG_PAGELENGTH 4
#define NTAG_BLOCKLENGTH 4*NTAG_PAGELENGTH
```
Note that CRC validation is omited in this example.

As for writing, have a look at this post: https://github.com/miguelbalboa/rfid/issues/197#issuecomment-325053383

CptAwe, 2018-05-27:
After 2 days of manipulating the library code, I still can't write data to an Ultralight card. Was anybody successful with this, or has an example?

I would really appreciate it!