miguelbalboa/rfid #133: Reading multiple cards

sigsevneo, 2015-10-31:
I''m facing a problem while reading multiple cards. The sketch is reading the first of the cards correctly, but fails to detect subsequent ones.

If I had to use the PICC function multiple times in one `loop()` to (read two different cards), how would I go about it?

My current code is something like:
```
lcd.clear();
lcd.setCursor(0,0);
lcd.print("READ CARD 1");      
      
// Look for new cards 
if ( ! mfrc522.PICC_IsNewCardPresent()) {//if PICC_IsNewCardPresent returns 1, a new card has been found and we continue
return;//if it did not find a new card is returns a '0' and we return to the start of the loop
}

// Select one of the cards
if ( ! mfrc522.PICC_ReadCardSerial()) {//if PICC_ReadCardSerial returns 1, the "uid" struct (see MFRC522.h lines 238-45)) contains the ID of the read card.
return;//if it returns a '0' something went wrong and we return to the start of the loop
}

lcd.clear();
lcd.setCursor(0,0);
lcd.print("READ CARD 2");
delay(3000);

// Look for new cards 
if ( ! mfrc522.PICC_IsNewCardPresent()) {//if PICC_IsNewCardPresent returns 1, a new card has been found and we continue
return;//if it did not find a new card is returns a '0' and we return to the start of the loop
}

// Select one of the cards
if ( ! mfrc522.PICC_ReadCardSerial()) {//if PICC_ReadCardSerial returns 1, the "uid" struct (see MFRC522.h lines 238-45)) contains the ID of the read card.
return;//if it returns a '0' something went wrong and we return to the start of the loop
}

```
The first two check conditions to check for a card work fine, but the next two don't function at all. Any ideas?

simbadMarino, 2015-10-31:
Hello Anagh Singh, in my personal experience I had to initialize again the
rfid module and it works to read several times both cards without removing
a placing the cards again in range of detection from RFID module.
I'm sharing a piece of code tha might help you... If you are removing and
placing the cards and it is not reading them again maybe it could be
something else...

#include <SPI.h>
#include <MFRC522.h>

#define RST_PIN 9 //
#define SS_PIN 7 //

byte buffer1, bufferSize =8, returnRequest;
byte uidType =4;

MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 instance

void setup() {
Serial.begin(9600); // Initialize serial communications with the PC
SPI.begin(); // Init SPI bus
mfrc522.PCD_Init(); // Init MFRC522
 // mfrc522.PCD_SetRegisterBitMask(mfrc522.RFCfgReg, (0x07<<4));
//Amplify signal
ShowReaderDetails(); // Show details of PCD - MFRC522 Card Reader details
Serial.println("Scan PICC to see UID, type, and data blocks...");
}

void loop() {

  mfrc522.PCD_Init();    // Init MFRC522

  delay(100);

// Look for new cards
if ( ! mfrc522.PICC_IsNewCardPresent()) {
return;
}
else
 {
 Serial.println("");
// Select one of the cards
if ( ! mfrc522.PICC_ReadCardSerial()) {
return;
}
/*returnRequest = mfrc522.PICC_RequestA(&buffer1,&bufferSize);
Serial.println(returnRequest);*/

//mfrc522.PICC_Select(&uidType,0);
 Serial.print("Card UID:");    //Dump UID
        for (byte i = 0; i < mfrc522.uid.size; i++) {
          Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
          Serial.print(mfrc522.uid.uidByte[i], HEX);
        }
// Dump debug info about the card; PICC_HaltA() is automatically called
// mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
//mfrc522.PICC_HaltA();

}

}
void ShowReaderDetails() {
// Get the MFRC522 software version
byte v = mfrc522.PCD_ReadRegister(mfrc522.VersionReg);
Serial.print("MFRC522 Software Version: 0x");
Serial.print(v, HEX);
if (v == 0x91)
Serial.print(" = v1.0");
else if (v == 0x92)
Serial.print(" = v2.0");
else
Serial.print(" (unknown)");
Serial.println("");
// When 0x00 or 0xFF is returned, communication probably failed
if ((v == 0x00) || (v == 0xFF)) {
Serial.println("WARNING: Communication failure, is the MFRC522 properly
connected?");
}
}

sigsevneo, 2015-11-01:
Thanks for the code. But as you mention I am removing it away from the range of detection. This code works in the use case which you specified, but unfortunately doesn't for mine.

mdxs, 2016-05-15:
@sigsevneo: check #40

closed 2016-08-25 by Rotzbua