miguelbalboa/rfid #135: ESP8266

asetyde, 2015-11-04:
I use your library with esp8266 nodemcu , but there are some small problems :  

- library advice about architecture not compatible
- some reboots
- sometimes codes can't uploaded , device busy
- not stable , reboot sometimes after card is read

linked issue : https://github.com/esp8266/Arduino/issues/963

Rotzbua, 2016-01-25:
I tested with a Wemos D1 mini. I had no issues, everything worked.

gutierrezps, 2016-04-17:
Although "Library advice about architecture not compatible" warning is issued, I'm using it with NodeMCU 1.0, and it's working fine with following pinout:

NodeMCU - RC522
D4 - RST
D5 - SCK
D6 - MISO
D7 - MOSI
D8 - SDA(SS)

"sometimes codes can't uploaded , device busy" I think it's Arduino, not this library.

"not stable , reboot sometimes after card is read" It's not happening here.

asetyde, 2016-04-17:
At now , with vers 2.0.0 or 2.1.0 i think ti can work , next week i will test it 
@Rotzbua it's a informatics  century 

@gutierrezps thanks i will report soon

asetyde, 2016-05-22:
Please add to read compatibility with ESP (it's not wright )

cpu bug resolved 
https://github.com/miguelbalboa/rfid/issues/212

closed 2016-05-22 by asetyde