miguelbalboa/rfid #138: MFRC522 & Arduino Nano

jpombas, 2015-11-18:
Hi,
I´m working with a RFID-RC522 module and Arduino Nano v3. When test with your example "firmware check" i got: 

*****************************
MFRC522 Digital self test
*****************************
Firmware Version: 0x92 = v2.0
Performing test...
-----------------------------
Result: OK

But when try reading a tag or card, the module does not recognize (detect) a TAG.

Can you help me? Sorry for my Engilsh :)

Rotzbua, 2015-11-20:
The firmware check only verify the firmware, as you see your firmware is ok. Also your wiring is ok, otherwise the check will fail.

So there are 2 possible problems:
1. TAGs are broken or not 13.56 MHz (NFC is not supported)
2. antenna of your module is broken, some people reported bad soldering at module

mdxs, 2016-05-15:
Unless @jpombas reports back, I recommend closing this issue assuming that hints from @Rotzbua helped resolve the issue.

closed 2016-05-15 by miguelbalboa