miguelbalboa/rfid #140: Software SPI & RFID

abhaysbharadwaj, 2015-11-28:
first of all, thank you for this wonderful library!

I have found that there are SoftwareSpi Libraries present. Any possibilities that software spi support can be integrated for this RFID library?

sophiekovalevsky, 2015-12-27:
What do you mean with Software SPI?

abhaysbharadwaj, 2015-12-28:
@sophiekovalevsky, Similar to arduino's "Software Serial" concept (i.e. any pin of the arduino can be programmed using 'software' (i.e. arduino code) to act as TX and RX) "Software SPI" means, any pin of the arduino can be programmed to act as the SPI pins (pins are: SDA, SCK, MOSI, MISO, RST).

abhaysbharadwaj, 2015-12-28:
@sophiekovalevsky, Similar to arduino's "Software Serial" concept (i.e. any pin of the arduino can be programmed using 'software' (i.e. arduino code) to act as TX and RX) "Software SPI" means, any pin of the arduino can be programmed to act as the SPI pins (pins are: SDA, SCK, MOSI, MISO, RST).

i found a good SoftwareSPI library, modified it and was able to attach my RFID to any pins of the arduino and use it. you can get the code at: www.github.com/abhaysbharadwaj/RFID-MRC522

P.S.: the normal SPI pins (a.k.a. Hardware SPI) are fixed pins on the arduino. i.e. pins 10,11,12,13 (www.arduino.cc/en/Reference/SPI) if you are using these pins for other peripherals, use software SPI to program other ping of arduino to behave like SPI pns and interface your sensor.

closed 2015-12-28 by abhaysbharadwaj