miguelbalboa/rfid #141: Overloading the constructor and init methode

lmmeng, 2015-11-29:
Hello Miki

Thanks for sharing your library, I'm using it in my [rfid2ln project] (https://github.com/lmmeng/rfid2ln).
After successfully used it for one MFRC522 reader connected to the Arduino, I'm trying to extend my project to handle more readers connected to the same Arduino. 
To make it easier to "automatically" handle different number of readers, I've overloaded the constructor with one without parameters (allowing me to define arrays of readers) and the PCD_Init() method with one doing the pins initialization: 
```code
void MFRC522::PCD_Init(byte chipSelectPin, byte resetPowerDownPin) {
	_chipSelectPin = chipSelectPin;
	_resetPowerDownPin = resetPowerDownPin;
	
	// Set the chipSelectPin as digital output, do not select the slave yet
	pinMode(_chipSelectPin, OUTPUT);
	digitalWrite(_chipSelectPin, HIGH);

	// Set the resetPowerDownPin as digital output, do not reset or power down.
	pinMode(_resetPowerDownPin, OUTPUT);
	// Set the resetPowerDownPin as digital output, do not reset or power down.
	pinMode(_resetPowerDownPin, OUTPUT);
```
Do you think you could implement these changes in your library? If yes, I can leave my project as it is, keeping your project as requirement. Otherwise I should also "distribute"  the changed library.

Many thanks,
Liviu

miguelbalboa, 2015-12-01:
Thank you @lmmeng , 
The idea of multiple readers is great, 
I will gladly to merge your changes if you open a pull request.

Greetings.

lmmeng, 2015-12-01:
Hello Miki,

I have just opened a pull request, so I'll close this issue.
It is the first pull request I've opened, I hope all things are OK.

Thanks,
Liviu

closed 2015-12-01 by lmmeng