miguelbalboa/rfid #145: example rfid_default_keys.ino authenticates with the first key only

mdanetzky, 2015-12-05:
If the PICC happens to be encrypted with other than first key from knownKeys array, the result is:
PCD_Authenticate() failed: Timeout in communication.
This code put in the line 151 seems to solve the issue:
```c
    while (true) {
      if (mfrc522.PICC_IsNewCardPresent() &&  mfrc522.PICC_ReadCardSerial()) {
        break;
      }
    }

mdxs, 2016-05-15:
I'm not sure how this would solve the claimed issue; but I also noted that the rfid_default_keys.ino example itself was updated on 5 Dec 2015... now is your proposed change applicable to before or after those changes?  It would be best to provide a Pull Request (PR) change proposal, or provide a little bit more context (as "line 151" is ambiguous...).

mdanetzky, 2016-05-16:
It is a very old issue and honestly I don't remember how this code solved the problem. I suggest to close it.

closed 2016-05-17 by miguelbalboa