miguelbalboa/rfid #149: How to get Rfid UID?

devishj, 2015-12-15:
Please can anyone suggest me a method on how to get RFID card UID just by scanning.

Rotzbua, 2015-12-15:
see example: rfid/examples/DumpInfo/DumpInfo.ino

devishj, 2015-12-15:
I have seen that. I want to extract UID from the information to be sent over to database. Please help me with this.

onokje, 2015-12-15:
I don't understand it either. I just want the ID. I tried:
Serial.println(mfrc522.PICC_ReadCardSerial());
But it allways prints 0

Rotzbua, 2015-12-15:
@onokje please read lookup the function before you complain, return type is bool so it has to be 0 or 1 ....
@devishj look at instructables there is a example with uid database doorlook system

hulkco, 2015-12-26:
Hi Devishj, maybe this code work for you:

//GeekChickens
//http://geekchickens.blogspot.com.es/
/*
SDA es el pin 10 Change to 6 for Share with Ethernet shield
SCK es el pin 13 Share with Ethernet shield
MOSI es el pin 11 Share with Ethernet shield
MISO es el pin 12 Share with Ethernet shield
RST es el pin 9 Change for 7 for Share with Ethernet shield
*/
//-----------------------------------------------variables, includes, funciones, etc-------------------------------------------------
#include <SPI.h>
#include <RFID.h>

#define SS_PIN 6
#define RST_PIN 7

RFID rfid(SS_PIN, RST_PIN);

int numero_serie[5]; // aqui guardaremos el número que nuestro sensor RFID detectará

 //-----------------------------------------------variables, includes,funciones, etc-------------------------------------------------

void setup()
{
  Serial.begin(9600);
  SPI.begin();
  rfid.init();
}

void loop()
{
   leer_usuario();
}

void leer_usuario()
{
    if (rfid.isCard())
    {
           if (rfid.readCardSerial())
            {
                Serial.print("Numero usuario: "); // guardamos el numero del usuario
                for(int i=0; i<=4 ; i++)
                {
                  numero_serie[i] = rfid.serNum[i];
                }

                for(int i=0; i<=4 ; i++) // y lo imprimimos
                {
                  Serial.print(numero_serie[i]);
                  Serial.print("    ");
                }

                Serial.println(" "); // espacio para el siguiente código
                delay(500); // para que no se imprima constantemente el mismo numero
             }
    }
    rfid.halt();
}

sophiekovalevsky, 2016-01-05:
@devishj there is a new example that you could use to accoplish that. 
See ReadNUID in examples folder.

Netoperz, 2016-01-26:
I'm here also with the same problem.
Should be a sample for getting UID as simple string in hex or decimal to send to database.
Right now we are getting bytes , byte after byte and we have to write a function to extract the UID etc. 

It should be just like in PN532 libraries a option in library to get it as string that can easily be able to send to another interface.

simbadMarino, 2016-06-13:
Hello Neto Perez, Serial.print is not adding the new line and Serial.println is indeed adding a new line per each increment in the loop, tats why you are seeing byte under byte, maybe you already figured this out because its an old post. I wanna help this Issue go to close. if you want to convert the readCard[] char variable into String just search in Google how to convert char to String in arduino, for sure you will find answers :)

closed 2016-08-25 by Rotzbua