miguelbalboa/rfid #165: arduino ide library support

Rotzbua, 2016-01-05:
Hi miguelbalboa,

maybe you like to support the arduino library manager.
To support this, the tags in the repository have to be also in semver format. This means `v1.1.3` are not allowed an have to be `1.1.3`.
Also every tag has to contain the right `library.properties`.

TODOs would be:
- rename tags to without `v`
- fix version in tag `1.1.2` in file `library.properties`

More details:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

miguelbalboa, 2016-01-06:
Hi @Rotzbua 
Thank you for the suggestion, I´ll do it.

Also Seems that we need to change the directory structure too as in https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#a-complete-example

Rotzbua, 2016-01-06:
Optionally. The format is backwards compatible to 1.0 and so the `src` folder is optional. You don't have to change the directory structure.
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#source-code

miguelbalboa, 2016-01-06:
I see, Thanks :)

closed 2016-01-06 by miguelbalboa

Rotzbua, 2016-01-11:
Hi miguelbalboa,
at tag 1.1.2 the version of library.properties is wrong, so the library manager wont work. Sadly the manager is not tolerant >:( ..
https://github.com/miguelbalboa/rfid/blob/1.1.2/library.properties

To fix this, something like this works:
```bash
git checkout 1.1.2
git branch patch-1.1.2
git checkout patch-1.1.2
** correct version library.properties to 1.1.2
git commit -m "bump version to 1.1.2"
git tag -f 1.1.2
git branch -D patch-1.1.2
git push -f --tags
```