miguelbalboa/rfid #168: UID of MasterCard PayPass / Visa Pay wave

Netoperz, 2016-01-11:
The RC522 should be able to read UID of any ISO14443 card

I'm looking for a solution to get UID of that atlas two kinds of banking cards.
I do not need any additional data just UID (no private personal banking data or whatsoever related ID)
The reader should be ale to get that ID as it is covered in ISO 14443-4 standard, at least that's the way it is identified by nfctools on samsung, where i can read that card without any problem.
Also PN532 reader and libnfc can read that info.

So could you add or point to a solution how to get that UID using Yours library ?

Rotzbua, 2016-01-11:
> The RC522 should be able to read UID of any ISO14443 card

That's wrong. Only ISO14443 **A**. Please read the readme...
> Datasheet:
> The MFRC522 reader supports ISO/IEC 14443 A/MIFARE mode.

norbertoramos, 2016-01-15:
I can't read this tag with the library. Why? The reader is RFID-RC522 0x92
It works on  RFID-RC522 0x91.
Thanks in advance and best regards.

1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  02  
       UID (NFCID1): af  c0  69  9c  
      SAK (SEL_RES): 38  
                ATS: 78  80  72  02  80  66  b0  07  01  01  07  07  53  02  31  10  82  90  00

Netoperz, 2016-01-15:
So, RC522 v1 0x91 can read Paywave and PayPas (visa/mastercard) UID , and v2 -0x92 can't ?
I'll check the version i have is it right.
I think that those cards are fit in to the ISO/IEC 14443A , and they have 4 byte UID as far as i can see so interesting is what is the difference 0x91 vs 0x92 .

Rotzbua, 2016-01-15:
I have two of version 0x92, both do no recognize the paypass card.
So I thought paypass uses nfc iso 18092, but you are right, they use iso 1443 A or B.
Source: https://www.paypass.com/pdf/EXTERNAL_PPMChip_Tmnl_Vendor_Testingv1.0_FINAL.pdf

I suggest that 0x92 has not so accurate timings as 0x91, compare checklist on page 71.

Netoperz, 2016-01-16:
I just checked the version i have, i think that they are "not exactly" made by NXP :)

ShowReaderDetails(); with a simple rule, shows 0x11 - unknown..

And... do not ask me why, they do read paypas, and paywave now.... 
but not with a dumpinfo sketch.

There was a problem reading paywave from one of banks, but after hardest now works fine.

i'll check the code what is the difference between mine and dumpinfo.
Strange things..... ;)

Netoperz, 2016-01-16:
That's the result:

1 read visa paywave
2 read visa paywave
3 read mastercard paypas

=======
MFRC522 Software Version: 0x11 (unknown)
Scan PICC to see UID, type, and data blocks...
Card UID: 97 21 45 57
PICC type: PICC compliant with ISO/IEC 14443-4
Dumping memory contents not implemented for that PICC type.

Card UID: 3F 42 88 1A
PICC type: Unknown type

Card UID: AF 2A C9 A2
PICC type: PICC compliant with ISO/IEC 14443-4
Dumping memory contents not implemented for that PICC type.

=====

So as you are also right, the second read cannot identify PICC type on the second Visa Paywave, and that card is also not dumped on two of three shields i have, all from one batch, all with the same chip, from the same production line...

No problems reading that card with different hardware, and with one of the shields ....
if the card was "B" type, should not be headed at all. so it must be A type, so why there ae some problems reading that particular card ?

I'm using dumpinfo now and it works as above.

Netoperz, 2016-01-16:
Ok, so i think i have solved the problem i a way... 
When i use 3,3v powering the arduino pro mini and the shield from FTDI 232r from macbook pro retina over USB. The reader does not want to read the cards other than mifare s20,s50.....

USB=>FTDI=>arduino pro mini VCCpin. + voltage regulator5vin to 3.3 => RC522 shield

also adding aside an external power source 2A 5v in place of USB/FTDI does not change nothing.
===
When i power up RC522 shield with 5v from FTDI/Arduino VCC line (not 3.3 as recommended) the reader can read most MasterCard paypas, and one visa paywave, the second one "that problematic" is sometimes dumped sometimes not, mostly not..

So when i use as a power source external 2A 5v charger, and i'm powering the RC522 with that, "automatically" the shield starts reading the "problematic" pay wave  without hesitation !!!!

So i think if you power up your shield with better power source it should start reading master card too :)

So i thing i have solved the mystery a bit, because if power source is the source of not reading/reading, so as far as i can assume , the only thing depending on power source that is related to that behavior is RF power.

So the question is can we change the RF power of rc522  ?
Maby instead of feeding it with to high (should be 3.3) 5v power, we can set the RF power to a higher value in a library, ad it will make reading banking cards more stable ?

What do you think ?

norbertoramos, 2016-01-16:
I'll try to overvoltage the shield.
Reading the bank cards with 0x91 version is done by reseting the module and then reading. I think this is a timing problem.
I'll make some tests.
Bests

Netoperz, 2016-01-16:
Thanks, check the RxGain_max, maby that will help, or tell me how and i'll check.
Would be good to have the solution soon, because i need to have working prototype for "yesterday", and I already have everything working flawlessly with PN532 using adafriut library with some tweaks.
But costs are 10 times more than using RC522, and I do not need all the emulation and p2p functions just for simple reading UID. But until we solve the problem with those cards i can't use RC522, over voltage the shield continuously may affect lifespan of IC, so the actual solution with over-voltage I have is just a temporary fix for further software solving the case. Im not a software developer, I do understand the code, but far far to say "i'm a developer" so my help here is limited to case solving and more hardware side solving as you can see :) 

I'm waiting for your suggestions to stabilize the banking cards reading.

Best regards

M.

norbertoramos, 2016-01-16:
I've supplied with 4.5V to the card and now it works.

Netoperz, 2016-01-16:
It's good then, because now You are able to look on the software side of the problem as you can read the  banking cards at all :).
If i can point to a most common problems, even with higher voltage there might be some problems with Some of the Visa Paywave , papas works well.
As I posted before, the smaller card's induction coil, the harder to read the card. with RC522 shield.
And that problem is most common on Visa.

So ones again i suggest to find a way to change the default RF power of the module, in both RX and TX, maby it will help ?

And what about timings you mentioned before ?

norbertoramos, 2016-01-16:
It seems that the problem is power in the card. In version 1 you must wait for or five seconds and then the card is read. Now I can read once a second but version 2 is what all we know.

Netoperz, 2016-01-17:
I'll check the NXP docs for both versions ones again and i'll try to contact someone in NXP with quotation about that situation.

Strange thing is, that in my fiends office they have one reader from china, no name, but i was able to open it, and there is PN522 and microchip microcobtroler plus bluetooth module, so the microcontroler is interfacing the NFC and BT, the NFC module is exactly the same i have on my boards, and that reader is using 4,7v from lion battery, and i suppose the microcontroler has got firmware written for that reader, and it reads the cards without problem. I had no time to exactly check the elements on the mainboard as the boss is a bit touchy when it comes to opening things with screwdriver :)

So it looks like it might be software dependent, and there might be a chance to improve the library, as the library was not tested with that cards because everyone was assuming that they cannot be read by c522 till i came here :)

norbertoramos, 2016-01-19:
I'll study the library to understand the problem with that reader and that cards.

Rotzbua, 2016-01-19:
My test results (reading UID):

PICC | PN532 | MFRC522 0x92
------------ | ------------ | ------------ 
MF classic - key cain | works | works
MF classic - card | works | works
MF ultralight - card | works | no reaction
MF ultralight - nfc tag| works | no reaction
paypass | works | no reaction

Both tested with arduino uno. Power supply by arduino, tested MFRC522 with 3.3V and 5V which make no difference with my module.

1. Detection distance with PN532 was 2cm, while MFRC522 it was 0.5cm. So maybe the transmitted power is too less to power the PICCs?

1. I found that version 2 has a additional timer, maybe this has an influence?
> The MFRC522 version 2.0 offers in addition a second prescaler timer. (datasheet p.32)

1. Another suggestion would be the communication modulation p.8 table 4. Maybe it works with a different setting.

Netoperz, 2016-01-19:
Please Add a 5V but external power to Arduino. IN my situation that have changed things.

USB-----------
                    |
                     ---------- VCC pin on arduino
                    |
EXTERNAL `5V

That will make reading the master card

If someone needs i can share, and send by post from Poland 2 RC522 shields with id 0x11

Just send me the address where to send, it might help with testing.

norbertoramos, 2016-01-22:
Hi,
I stated that you need a RF power amplifier to work with paypass and paywave.

norbertoramos, 2016-01-22:
http://cache.nxp.com/documents/application_note/AN1445_An1444.zip?fpsp=1&WT_TYPE=Application%20Notes&WT_VENDOR=FREESCALE&WT_FILE_FORMAT=zip&WT_ASSET=Documentation&fileExt=.zip
Contiene algo de información. Adjunto un resumen donde se ve que es necesario el Booster de RF para las tarjetas bancarias.

norbertoramos, 2016-01-22:
Sorry, I spoke in spanish.
That link has some information. Find enclosed a documento of NXP with info about need of booster.
[75016623.pdf](https://github.com/miguelbalboa/rfid/files/100573/75016623.pdf)

Netoperz, 2016-01-25:
[image]

This is block diagram.
But it refers to a PN523 (that module works without it like a charm, so no idea why it is there)
Right now with a bit bigger power 4,5-5V we can achieve the same.
But when the mifare cards can be read from 5 cm in my case, than the master card and papas from 0,5, what is quite fine. I think it is due to the "anti fraud" security those cards are designed to be really near the reader, to prevent scanning them from longer distance .

That amplifier can be used as methods for reading those cards from longer distances.

What about that option:

Table 98. RFCfgReg register bit descriptions
Bit Symbol Value Description
7 reserved - reserved for future use
6 to 4 RxGain
[2:0]
defines the receiver's signal voltage gain factor:
000 18 dB
001 23 dB
010 18 dB
011 23 dB
100 33 dB
101 38 dB
110 43 dB
111 48 dB
3 to 0 reserved - reserved for future use

So did anyone improved the range by changing the gain to 48 dB ?

It must be done by modification of the library or in the sketch ? and if in the sketch, than how ?

Rotzbua, 2016-01-25:
It is already in the library, but you have to set in manually to max, because it may damage the chip.
The readme mention these option to increase antenna gain.

Netoperz, 2016-01-26:
So i see that nothing more than just overpowering or using a different module, or redesigning and etching from scratch the module with additional ampler can solve the problem.

So my conclusion is that RC522 is a module with a very limited functionality. That can read only some cards without problem, and as you can never be sure what version you are ordering (all made in china) if you expect building a device within the specs, without, overpowering, without rebuilding the reader module from scratch, you should use PN532 module, that module costs a bit more- 5-8$ with shipping from china, but works with more types of cards, works with A and B and has got a lot more functions, like emulation of tag, P2P, fully compatible with NFC standard.

And also the libraries from adafriut and other vendors are in public domain as GPL, so RC522  is a solution only if you need just a simple one type of card working, and you need to save 3,5$ (because that is the difference in price of RC522  and PN532)

I'm closing the issue.

closed 2016-01-26 by Netoperz

hydra228, 2018-02-06:
bro use 5v PIN rc522 on arduino! all works)