miguelbalboa/rfid #172: Read/Write codes..Error in Compiling

jhed20, 2016-01-16:
i copy and paste the Read/Write code of the RFID examples and i got many error.. i dont know why.. can someone please help me this problem.. sorry, im new to the arduino world speacilly the RFIDS.. thank you.. respect :D
[image]

lmmeng, 2016-01-16:
Hello @jhed20,
you should install the MFRC522 library before using it - see https://www.arduino.cc/en/Guide/Libraries for details about installing libraries in Arduino. 
After installing the library you don't need to copy/paste any code, you'll find it in the examples (File -> Examples -> MFRC522).

jhed20, 2016-01-16:
i actually done it sir.. i dont know why is that happen.. attached here is
​the picture after installing the library

jhed20, 2016-01-16:
when i compile the codes given in the example the Read/Write then i got this error

[image]

jhed20, 2016-01-17:
i try to use the Arduino v1.6.8 which is the latest version of arduino and
install the library and it works fine when i verify the code but when i
upload it then i've got an error again.. this is my error sir

processing.app.debug.RunnerException
at
cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:179)
at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
at processing.app.Sketch.upload(Sketch.java:1178)
at processing.app.Sketch.exportApplet(Sketch.java:1152)
at processing.app.Sketch.exportApplet(Sketch.java:1124)
at processing.app.Editor$DefaultExportHandler.run(Editor.java:2417)
at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NullPointerException
at java.lang.String.replace(String.java:2240)
at
processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:97)
at
processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:89)
at
processing.app.helpers.StringReplacer.formatAndSplit(StringReplacer.java:37)
at
cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:174)
... 6 more

lmmeng, 2016-01-17:
Have you choose the right com port (Tools -> Port) for your Arduino board?

jhed20, 2016-01-18:
yes sir.. i did it already.. still ive got that error, so i switch to arduino v 1.6.6 and it works ok.. theres no problem in compiling but my problem now is, it wont work.. when i run the serial monitor i only got this message,

"Scan a MIFARE Classic PICC to demonstrate read and write.
Using key (for A and B): FF FF FF FF FF FF
BEWARE: Data will be written to the PICC, in sector #1
"

and when i scan my id tag nothings happen.. why ?.. please help me.. thank you for ur response sir Immeng

jhed20, 2016-01-18:
heres the screenshot picture sir...

[image]

Merry27, 2016-01-18:
hello sir Immeng,. what version of arduino did you used sir?
Thanks ahead sir .

lmmeng, 2016-01-18:
I'm using Arduino 1.6.5, but I don't think the problems are related to the Arduino. It looks more as hardware problem (SPI communication not working).
Without any details about you hardware is quite difficult to debug it, but as suggestion check your hardware connections and the compatibility of the reader/tags with the list in the comments in the library.

Merry27, 2016-01-18:
got it , thanks a lot sir.  :)

jhed20, 2016-01-18:
maybe your right sir that the problems are not related to the arduino. Anyway, i reconnect the connections a while ago and i got this problem .. 

Scan a MIFARE Classic PICC to demonstrate read and write.
Using key (for A and B): FF FF FF FF FF FF
BEWARE: Data will be written to the PICC, in sector #1
Card UID: 8C 3C 04 C5
PICC type: MIFARE 1KB
Authenticating using key A...
Current data in sector:
   1      7   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
          6   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
          5  MIFARE_Read() failed: Timeout in communication.
          4  MIFARE_Read() failed: Timeout in communication.

Reading data from block 4 ...
MIFARE_Read() failed: Timeout in communication.
Data in block 4:
 30 04 26 EE 01 0A 00 02 00 00 23 A3 00 10 EA 00

Authenticating again using key B...
PCD_Authenticate() failed: Timeout in communication.

Merry27, 2016-01-18:
thanks a lot sir,. btw ,me and Jhed20 are friends and were helping each
other to solve this problem.

when he scan the ID tag/card the serial monitor shows ,  failed: Timeout in
communication.

jhed20, 2016-01-18:
[image]

mdxs, 2016-05-15:
Issue can be closed, as I would assume that the friends have resolved their problems by now (which started with an older Arduino IDE and later seem to indicate that before reading they jumped into writing right away... I would recommend starting with just reading stuff first).

closed 2016-05-15 by miguelbalboa

tirumalagithub, 2016-10-27:
hi every one i am getting same error but i have tried lot of times by reconnecting the hardware ....still it is getting same error ....actually i am using arduino mega board...can any one help me out