miguelbalboa/rfid #173: Question about reading from resetPowerDownPin

krayushkin, 2016-01-16:
I ported the library in С and have a question about this code:
```c
	void MFRC522::PCD_Init() {
	// Set the chipSelectPin as digital output, do not select the slave yet
	pinMode(_chipSelectPin, OUTPUT);
	digitalWrite(_chipSelectPin, HIGH);
	
	// Set the resetPowerDownPin as digital output, do not reset or power down.
	pinMode(_resetPowerDownPin, OUTPUT);
	if (digitalRead(_resetPowerDownPin) == LOW) {	//The MFRC522 chip is in power down mode.
		digitalWrite(_resetPowerDownPin, HIGH);		// Exit power down mode. This triggers a hard reset.
		delay(50);
	}
	else { // Perform a soft reset
		PCD_Reset();
	}
```
We do digitalRead(_resetPowerDownPin) and compare result, but earler we set this pin to push pull OUTPUT and by default it will be LOW. I think, we have dead code here. Is it true?

sophiekovalevsky, 2016-01-17:
Why we should have a default value when we declare pinMode()?
That functions just only set the I/O pin as a OUTPUT or and INPUT there is no default value as far as I know.

krayushkin, 2016-01-17:
Thanks, I understand, second branch executed on second and later calls MFRC522::LCD_Init() function. It's my mistake.

closed 2016-01-17 by krayushkin