miguelbalboa/rfid #175: Firmware Version: 0x50 = (unknown)

sarri77, 2016-01-19:
My RC522 doesn't work!!

When I push your examples it seems everything it's ok, but it doesn't read any card.

I received a strage message:

"Firmware Version: 0x50 = (unknown)"

It seems like the firmware is not correct, can you help me?

mdxs, 2016-05-15:
Not without further information. Please note that you only provided information on what doesn't work, but no details on: what you used (all the hardware), how you connected it (the wiring) and what you tried (which of the examples).

KC38, 2016-08-04:
Hi,
I am new to RFID.
I used Arduino Uno, RFID reader RC522 and did the same connections as it is in the code. It still does not work. 

Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?

These two error messages are shown in the serial monitor. Please help

closed 2016-08-25 by Rotzbua