miguelbalboa/rfid #177: NFC-Chip: Broadcom Topaz512

eferdi, 2016-01-20:
Hi,
I already connected the reader to the arduino and could read Mifare Classic 1k and Ultralight.
But now I got some Broadcom Topaz512 Chips and I can not convince the library to read them.
Is it even possible?

NFC-Chip: Broadcom Topaz512
Memory: 512 Byte (usable: 456 Byte)
Frequenz: HF 13.56MHz
Standards: ISO 14 443-2 A, ISO 14 443-3 A | NDEF formatted

Rotzbua, 2016-01-21:
Yes it should be possible. Datasheet says it's compatible to ISO14443A. By this library there is no direct support for those chips, you have to implement it. So actually the chips should be recognized as unknown or wrong type.

Rotzbua, 2016-08-25:
@soulflyman  Hi. Did you found a solution? Or a more detailed bug description?
Otherwise I will close this issue.

Rotzbua, 2016-09-24:
no response -> close

closed 2016-09-24 by Rotzbua