miguelbalboa/rfid #178: SPI issue rc-522 + nrf24l01

alexandre72, 2016-01-21:
I'm trying to use rfid library + RF24 lib from https://github.com/TMRh20/RF24 without any success. I don't see where's the problem.

rc522 and nrf24l01 share MOSI, MISO and SCK lines
rc522  RST on pin 7 and SS on pin 8
nrf24l01 CE on pin 9 and CSN on pin 10
 Here's sample code:

  //nrf24l01 RF24
  SPI.begin();
  radio.begin();
  radio.setPALevel(RF24_PA_MAX);
  radio.printDetails();

  // RFID MFRC522
  rfid_last_time = 0;
  mfrc522.PCD_Init();
  mfrc522.PCD_DumpVersionToSerial();
  radio.printDetails();

sometimes rc522 works, it can read tag but block nrf24l01 and sometimes  nrf24l01 blocks rc522

Rotzbua, 2016-01-25:
Maybe the SPI select (SS/CSN) is not set correctly. 
The rfid library should work correctly, so my hint would be to check the CSN code in the rf24 library.
Best

alexandre72, 2016-01-30:
Thanks for answer! solved(problem wasn't related to RF24 or RFID lib). Sorry :-1:

closed 2016-01-30 by alexandre72

jpalmowski1, 2016-02-26:
alexandre72, what solved your problem? I can't seem to get these two to work together either.

alexandre72, 2016-02-26:
I used sofware RFID library, all is ok now

jpalmowski1, 2016-02-26:
can I see your arduino code? Also did you use the 3.3V from the arduino to
power the RF24?

arpitjayabt, 2017-09-09:
Hello alexandre72 please tell me what solved your problem. I can't make it work together. Please can you send me your code. I am really stuck.
arpitjayabt@gmail.com

plc-ecy, 2024-12-10:
hi fellas, i have some problem...so can i learn the solution??i need to learn which library used..