miguelbalboa/rfid #192: Write only one bytes on a MIFARE Ultralight

brunosso, 2016-02-19:
I do some test with writing page 2 on a MIFARE Ultralight, editing the examples ReadAndWrite. I see that the function MIFARE_Write use 16 bytes but only the least significant 4 bytes (bytes 0 to 3) are written. So i edit the array **dataBlock[]** and i successfuly write the bytes i need, that is the byte 2.
`  byte dataBlock[]    = {
    0x8a, 0x48, **0xfa**, 0x03,
    0x3f, 0xff, 0xff, 0xfe,
    0x01, 0x04, 0x00, 0x00,
    0x02, 0x01, 0x02, 0xbe,
  };`

Now i want to write only the byte 2 (**0xfa** in this case) regardless of 0,1,3... is possible?

Rotzbua, 2016-02-19:
No, always blocks are read or written. Additionally in- and decrement commands are available.

brunosso, 2016-02-19:
an idea, is to update the datablock[] array, with a dump of read value? what do you think?

Rotzbua, 2016-02-21:
Yes, if you only want to manipulate one byte, you first read the block then modify the byte and write it back to the PICC.

If you only want to store one byte and you dont use the other bytes I would simply write the byte multiple into the block. So you have additional data redundancy = data safety.

mdxs, 2016-05-15:
Can be closed, assuming @brunosso solved his issue with the advise from @Rotzbua.

closed 2016-05-15 by miguelbalboa