miguelbalboa/rfid #193: Making RC522 work with Teensy 3.x

sdierick, 2016-02-20:
Hello,
I've read several forum entries about this lib not being compatible with 32bits MCU like the one embeeded with Teensy 3.x.
When migrating from arduino to teensy on my project , I didn't manage to get my RC522 to work with the lib.
If someone can explain me the problem with 32bits and this lib, I can try a fork to get it work. Several libs are now available for Teensy and Teensy is compatible with Arduino IDE and much more powerful than arduino! Than could be great to get it done.

Rgds ans thx for this wonderful lib!

Rotzbua, 2016-02-21:
duplicate #57

Rotzbua, 2016-02-21:
The common issue for ports would be the SPI.

If you find the issue, I would prefer preprocessor statements for compatibility instead of a extra fork.

I found a older smaller code, seems modified for 32bit teensy:
https://github.com/sumotoy/RFID/tree/master

sdierick, 2016-02-21:
Thanks for your reply. I'm going try to sort things out. I'll keep it as preprocessing statement if possible. 
I've read on the other issue that the problem was probably a pointer issue because of 8bits adresses being initialized and passed as a pointer. What about the SPI issue you are talking about ?

Rotzbua, 2016-02-22:
I suggested:
1. The libraries for SPI are sometimes different.
2. SPI can use different speed and mod settings. Not every architecture supports all combinations. So this can be a point of failure.

I do not think it is a problem of 32bit, I successfully tested the library on a esp8266 and it has a 32bit risc processor.

sdierick, 2016-03-01:
Hello All,

I've tried the sumotoy old RF522 Lib... No way...
I've tried to put some Serial.println inside to figure out what happens... It seems that I'm always reading FF from any RC522 registrer... Which is obviously not what we want...
I've tried to change SPI mode, spi speed from this library with no success.

On the other hand, I also have to work with teensy and an BTLE adafuit module also on SPI and which worked like a charmed without any lib modification out of the box...

Sorry guys, but it seems I've reached the bottom of my (short) knowledge as a embedded developper... Any help would be appreciated after those hours spent in the desert...

Cheers.

PaulStoffregen, 2016-03-01:
I tested today with Teensy 3.2.  It seems to work, at least with the DumpInfo example.

https://forum.pjrc.com/threads/25596-Teency-3-1-Compatibility-with-MFRC522-RFID-Library?p=97871&viewfull=1#post97871

[image]

Rotzbua, 2016-08-25:
@sdierick Hi. Did you found a solution? Otherwise I will close this issue.

Rotzbua, 2016-09-24:
no response -> close

closed 2016-09-24 by Rotzbua