miguelbalboa/rfid #194: How do other things while waiting a card?

samuelrcarvalho, 2016-02-21:
Hi there.

I´m using this code..
`MFRC522::StatusCode status;
status = (MFRC522::StatusCode) mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, 1, &key, &(mfrc522.uid));
status = (MFRC522::StatusCode) mfrc522.MIFARE_Read(1, buffer, &size);`

And after that, the arduino is waiting a card for read.

So, I´m using a 7 seg display, and while the reader is waiting, my arduino should do others things, like to calculate and show things in this display. But due this waiting time all is stopped.
how can I do more things futher waiting a card?
Is there a timeout for MIFARE_Read?

thanks

Rotzbua, 2016-02-22:
Hi.
Only possible with interrupts. The library do not support this method, you have to implement this.

A proof of concept exists:
https://github.com/miguelbalboa/rfid/blob/master/examples/MinimalInterrupt/MinimalInterrupt.ino

samuelrcarvalho, 2016-02-22:
Ok Rotz.

Thank you for your attention.

closed 2016-02-22 by samuelrcarvalho

Rotzbua, 2016-02-22:
But reading a picc only takes less than a second, so there is not much time to calc other stuff.

May use following helper fcts before you auth and read, so you only auth if picc is present.
```
	bool PICC_IsNewCardPresent();
	bool PICC_ReadCardSerial();
```