miguelbalboa/rfid #196: Arduino pro micro (Leonardo)  Atmega32u4  Serial corupted data

Netoperz, 2016-03-19:
Hello,

Arduino IDE 1.6.6, 1.6.7
RC522 library 1.6 .17.1.8 ...

code:

[code]
#include <SPI.h>
#include <MFRC522.h>

#define RST_PIN         9
#define SS_PIN          10
        MFRC522 mfrc522(SS_PIN, RST_PIN);
        
void setup() {
  Serial1.begin(9600);
  SPI.begin();
    mfrc522.PCD_Init();
}

void loop() {
    if ( ! mfrc522.PICC_IsNewCardPresent())
        return;
    if ( ! mfrc522.PICC_ReadCardSerial())
        return;  
        
           Bluetooth_HID(mfrc522.uid.uidByte, mfrc522.uid.size);           
          
       mfrc522.PICC_HaltA();
       mfrc522.PCD_StopCrypto1(); 
}

void Bluetooth_HID(byte *buffer, byte bufferSize) { 
   for (byte i = 0; i < bufferSize; i++) {
        Serial1.print(buffer[i] < 0x10 ? "0" : "");
        Serial1.print(buffer[i], HEX);
    }
} 
[/code]

When i upload the code i got proper UUID , when i unplug the board from USB (power) and plug it back again i have got garbage on serial when reading the UUID, exatly in hex 80 80 80 80 80 80 80, doesn't matther what card i'm using.

Problem is showing after disconecting the powersource to the Arduino, before that, when you are reseting the board and just after uploading the code, the code works fine.

The problem occured now, and the same sketch flashed aboud 3 weeks ago when i was programming readers is working well.

When i flash it now to the device that is working well after flashing and reconecting the power it has the problem as described.

Any idea what is going on  ?
Can anyone test it on similar board with rc522 v2 shield on ? i was testing the sketch with software serial to pins 3 and 2 and the same problem.

Everything works untill you diconnect the board from usb and plug it back, and then you got garbage, corupted data on serial.

Rotzbua, 2016-03-22:
If I remember correctly: the atmega32u4 is different to the uno: Serial and USB use the same pins

Try from examples:
```
 while (!Serial); // Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)

Netoperz, 2016-03-22:
Problem fixed in a way.... i just burned boot loader to the boards, and everything was back normal.

I think that new Arduino IDE is messing with memory that may in some way influence the boot loader, or maybe some trash is staying in rom while flashing new software over and over.

It is hard to say, but for future info if someone will encounter something like me. Just stop bothering with why how when where, whatdaf.......k.... etc...

Just burn fresh boot loader flash your sketch and you are good to go :)

Thx for reply but i have managed to solve the problem in that way.

Cheers, case closed.

closed 2016-03-22 by Netoperz