miguelbalboa/rfid #197: Write to NTAG216?

EdgyLedgyWedgbarge, 2016-03-20:
Hello. I've just purchased a C522 RFID Reader and I been able to read / write to Mifare classic tags with my Arduino Uno without any problems.
What I'd really like to be able to do is to read / write to NTAG216 tags. I'm able to read the data but I'm not able to write. 
Any ideas how I can achieve this?

Rotzbua, 2016-03-22:
I think NTAG216 tags may use other commands for writing data than mifare. The data sheet will tell you more.

mdxs, 2016-05-15:
And when @EdgyLedgyWedgbarge does find how to do this, I'm sure a Pull Request (PR) would be welcome ;-)

leandre84, 2017-08-25:
I'm able to write to NTAG 213/216 using MIFARE_Ultralight_Write() without any issues. You just need an 4x8 bit array to hold the data for each page to write, have a look at the following code I use to write pages 4 and 5 with factory data according to the NTAG datasheet as an example

`int NTAGWriteDefaultUserPages() {
	int status = 0;
	byte page4[NTAG_PAGELENGTH];
	byte page5[NTAG_PAGELENGTH];

	#if (CARD_TYPE == 203)
	page4[0] = 0x01; page4[1] = 0x03; page4[2] = 0xA0; page4[3] = 0x0C;
	page5[0] = 0x34; page5[1] = 0x03; page5[2] = 0x00; page5[3] = 0xFE;
	#endif
	#if (CARD_TYPE == 206)
	page4[0] = 0x03; page4[1] = 0x00; page4[2] = 0xFE; page4[3] = 0x00;
	page5[0] = 0x00; page5[1] = 0x00; page5[2] = 0x00; page5[3] = 0x00;
	#endif
	status += mfrc522.MIFARE_Ultralight_Write(4, page4, NTAG_PAGELENGTH);
	status += mfrc522.MIFARE_Ultralight_Write(5, page5, NTAG_PAGELENGTH);

	return status;
}`

kazezx, 2018-02-24:
how do i use the Read/Write for Ultralight Tag Ntag213 ? i get timeout error when i try using the library.

Rotzbua, 2020-07-25:
Solution from leandre84 moved to wiki https://github.com/miguelbalboa/rfid/wiki/Tag-NTAG-213-216

closed 2020-07-25 by Rotzbua