miguelbalboa/rfid #199: some example doesn't work with MFRC522 firmware v2.0

yaroslavVl, 2016-03-22:
MFRC522 Firmware Version: 0x92 = v2.0
rfid-master v1.1.8
cards (i've tried 3 cards): PICC type: MIFARE 1KB Card SAK: 08 (or 88)

examples: RFID-Cloner, rfid_default_keys.

problem: it works only if the first key is right.
But if I chage first key to wrong and change second (or other else) to right, the program try it, but it doesn't work (it is look like incorrect key).

How can I reconnect to card without take out it(card)?
	
[image]

mdxs, 2016-05-15:
Don't understand exactly... but is this perhaps an A and B key issue?  @yaroslavVl: please read the example code before trying it out; especially if it shows BEWARE or WARNING messages; as it might change how sections can be read after the change.

Rotzbua, 2016-08-25:
@yaroslavVl Hi. Did you found a solution? Or a more detailed bug description?
Otherwise I will close this issue.

Rotzbua, 2016-09-24:
no response -> close

closed 2016-09-24 by Rotzbua