miguelbalboa/rfid #200: Error while executing Dumpinfo code on raspbian

howdoidoit, 2016-03-23:
I am trying to run this code in arduino IDE that I have installed on rasberry pi. I have included MFRC522 library in arduino IDE that I generated using make command after extracting the zip folder downloaded from here. When I try to compile the code, it shows following errors. Please help..  

/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp: In member function â€˜void MFRC522::PCD_WriteRegister(byte, byte)â€™:
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:41:6: error: â€˜class SPIClassâ€™ has no member named â€˜beginTransactionâ€™
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:41:70: error: â€˜SPISettingsâ€™ was not declared in this scope
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
                                                                      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:46:6: error: â€˜class SPIClassâ€™ has no member named â€˜endTransactionâ€™
  SPI.endTransaction(); // Stop using the SPI bus
      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp: In member function â€˜void MFRC522::PCD_WriteRegister(byte, byte, byte*)â€™:
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:57:6: error: â€˜class SPIClassâ€™ has no member named â€˜beginTransactionâ€™
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:57:70: error: â€˜SPISettingsâ€™ was not declared in this scope
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
                                                                      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:64:6: error: â€˜class SPIClassâ€™ has no member named â€˜endTransactionâ€™
  SPI.endTransaction(); // Stop using the SPI bus
      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp: In member function â€˜byte MFRC522::PCD_ReadRegister(byte)â€™:
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:74:6: error: â€˜class SPIClassâ€™ has no member named â€˜beginTransactionâ€™
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:74:70: error: â€˜SPISettingsâ€™ was not declared in this scope
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
                                                                      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:79:6: error: â€˜class SPIClassâ€™ has no member named â€˜endTransactionâ€™
  SPI.endTransaction(); // Stop using the SPI bus
      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp: In member function â€˜void MFRC522::PCD_ReadRegister(byte, byte, byte*, byte)â€™:
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:98:6: error: â€˜class SPIClassâ€™ has no member named â€˜beginTransactionâ€™
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:98:70: error: â€˜SPISettingsâ€™ was not declared in this scope
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
                                                                      ^
/home/pi/sketchbook/libraries/MFRC522/MFRC522.cpp:121:6: error: â€˜class SPIClassâ€™ has no member named â€˜endTransactionâ€™
  SPI.endTransaction(); // Stop using the SPI bus
      ^

Rotzbua, 2016-03-23:
Something is wrong with your SPI lib.
Which ide version and board did you tried?

howdoidoit, 2016-03-23:
arduino 2:1.0.5+dfsg2-4 IDE. I installed it using '_sudo apt-get install arduino_' command
and I am using arduino uno board
can I install new version of IDE or can I upgrade SPI library?

Rotzbua, 2016-03-23:
Ok, 1.0 is too old, this library targets the latest 1.6.
Maybe upgrading SPI lib works but I don't know.

howdoidoit, 2016-03-23:
Yes, 1.6 uses new version !
But 1.0 version was installed by default from command line.
Can you please suggest how can I install a newer version?

howdoidoit, 2016-03-23:
Or how can I upgrade SPI lib?

Rotzbua, 2016-03-23:
The repo do not provide a newer version and website do not offer a arm compiled version. So you have two options: 1. get source code and compile it by your own or 2. search a arm compiled version on the internet.

To upgrade SPI lib: search spi.h and spi.cpp and replace them by the latest source.

howdoidoit, 2016-03-23:
Okay!
Thank you for helping :-)

mdxs, 2016-05-15:
Issue can be closed.

commit e9003130 (2016-08-25): upd readme

add note for compatible ide
closes #200
closes #204
closes #205
closes #212
upd readme
closes #220
closes #36
extend gitignore with ide clion data
closed 2016-05-15 by miguelbalboa