miguelbalboa/rfid #201: My arduino doesnt want to upload the code

thaama, 2016-03-26:
help me i am trying to upload dump info to my arduino module but it always show this code
Arduino: 1.6.8 (Windows 8.1), Board: "Arduino/Genuino Uno"

Sketch uses 9,574 bytes (29%) of program storage space. Maximum is 32,256 bytes.
Global variables use 286 bytes (13%) of dynamic memory, leaving 1,762 bytes for local variables. Maximum is 2,048 bytes.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x8a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x8a

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

i am using arduino uno and mifare rc552

eferdi, 2016-03-26:
Had the same problem with my linux laptop and arduino uno. Luckily a reboot solft the problem for me.

simbadMarino, 2016-03-26:
Same here, if you are using Linux a few reboots may fix it, or at least it did in my case. If you are using Windows make sure you are using the correct arduino versión in the ide and also the right programmer. But usully a reboot should do the trick.
In case none of this helps I would recommend reinstalling the drivers and downloading again the ide and start from 0. Best regards and I hope you solve it :)
--
老的不去，新的不来… sábado,

Rotzbua, 2016-03-26:
The issue is wrong here, plz report it to the arduino ide repo: https://github.com/arduino/Arduino/issues

mdxs, 2016-05-15:
I agree that this issue doesn't belong here and thus should be closed.

closed 2016-05-15 by miguelbalboa