miguelbalboa/rfid #204: unable to compile with arduino 1.0.4

happyuser007, 2016-04-13:
I have imported the library and tried to run dumpinfo.ino. When I compile, I get several warnings saying "C:\arduino-1.0.4\libraries\rfid_master/MFRC522.h:123: warning: only initialized variables can be placed into program memory area"
and an error that aborts the process 
"C:\arduino-1.0.4\libraries\rfid_master/MFRC522.h:276: error: use of enum 'PICC_Type' without previous declaration"
it seems that the other enum declarations are OK but that one is typed (byte type) and it is not accepted.
Have you ever seen such behaviour and what do you suggest ? Thank you.

Rotzbua, 2016-04-14:
Arduino IDE 1.0 use a old c++ compiler, try latest version.

happyuser007, 2016-04-14:
Do you mean latest version of C++ compiler? I tried the latest Arduino version (1.6.8) and it's the same. Should I replace the compiler inside arduino tools, because I am not used to C++ compiler and dont know how to compile and link..

Rotzbua, 2016-04-14:
I meant the latest arduino ide. It supports c++11 which is used for the enum declaration.

If you do not find the error, simply change c++11 to old-style c++ back by removing `: byte`.

mdxs, 2016-05-15:
I think this issue should be closed; as @happyuser007 should use a more recent installation of the Arduino package.

commit e9003130 (2016-08-25): upd readme

add note for compatible ide
closes #200
closes #204
closes #205
closes #212
upd readme
closes #220
closes #36
extend gitignore with ide clion data
closed 2016-05-15 by miguelbalboa