miguelbalboa/rfid #207: RFID compile with Arudino MEGA for read/write & dumpinfo

kendrickchin, 2016-04-22:
Hi, i used the code provided below and i couldn't compile ReadAndWrite due to an error,
 "invalid conversion from 'byte {aka unsigned char}' to 'MFRC522::PICC_Type' [-fpermissive]". 

[image]

As for Dumpinfo, the error is "'class MFRC522' has no member named 'PCD_DumpVersionToSerial'"

[image]

Rotzbua, 2016-04-22:
Update your mfrc522 lib. I added it with commit f7f2d5d02dd545c4b27c855219437d2712d61434 .

kendrickchin, 2016-04-22:
how do i update?

Rotzbua, 2016-04-22:
https://www.arduino.cc/en/Guide/Libraries#toc3
or replace the files

kendrickchin, 2016-04-22:
I cant find the download button for https://github.com/miguelbalboa/rfid/blob/master/MFRC522.cpp

closed 2016-04-22 by kendrickchin