miguelbalboa/rfid #208: PCD Authentication() failed: Timeout in communication.

kendrickchin, 2016-04-22:
Hi, I tired on an empty card and for dumpinfo, it shows all the sector however, when i try reading an used card, on the last 3 sectors, it shows PCD Authentication() failed: Timeout in communication. Why is it so?

[image]

Rotzbua, 2016-05-03:
Bad communication between tag and reader. It is normal.
Do not move the tag while reading. Maybe increase baud to decrease read time.

@miguelbalboa Can be closed.

mdxs, 2016-05-15:
Agree that this can be closed.

closed 2016-05-15 by miguelbalboa

hadipic, 2017-09-19:
uint8_t PCD_Authenticate(spi_device_handle_t spi,
                                            byte command,		///< PICC_CMD_MF_AUTH_KEY_A or PICC_CMD_MF_AUTH_KEY_B
											byte blockAddr, 	///< The block number. See numbering in the comments in the .h file.
											MIFARE_Key *key,	///< Pointer to the Crypteo1 key to use (6 bytes)
											Uid *uid			///< Pointer to Uid struct. The first 4 bytes of the UID is used.
											) {
	byte waitIRq = 0x10;		// IdleIRq

	// Build command buffer
	byte sendData[12];
	sendData[0] = command;
	sendData[1] = blockAddr;
	for (byte i = 0; i < 6; i++) {	// 6 key bytes
		sendData[2+i] = key->keyByte[i];
	}
	// Use the last uid bytes as specified in http://cache.nxp.com/documents/application_note/AN10927.pdf
	// section 3.2.5 "MIFARE Classic Authentication".
	// The only missed case is the MF1Sxxxx shortcut activation,
	// but it requires cascade tag (CT) byte, that is not part of uid.
	for (byte i = 0; i < 4; i++) {				// The last 4 bytes of the UID
		sendData[8+i] = uid->uidByte[i+uid->size-4];
	}

    uint8_t backData[16];

  	// Start the authentication.
	return PCD_CommunicateWithPICC(spi,PCD_MFAuthent, waitIRq, &sendData[0], sizeof(sendData),backData, sizeof(backData), 0,0,0);
} // End PCD_Authenticate()

hadipic, 2017-09-19:
i convert c++ rotin to c but not worcing
help

hadipic, 2017-09-19:
uint8_t PCD_CommunicateWithPICC(spi_device_handle_t spi,
                                                        uint8_t command,       ///< The command to execute. One of the PCD_Command enums.
                                                        uint8_t waitIRq,       ///< The bits in the ComIrqReg register that signals successful completion of the command.
                                                        uint8_t *sendData,     ///< Pointer to the data to transfer to the FIFO.
                                                        uint8_t sendLen,       ///< Number of bytes to transfer to the FIFO.
                                                        uint8_t *backData,     ///< nullptr or pointer to buffer if data should be read back after executing the command.
                                                        uint8_t *backLen,      ///< In: Max number of bytess to write to *backData. Out: The number of uint8_ts returned.
                                                        uint8_t *validBits,    ///< In/Out: The number of valid bits in the last byte. 0 for 8 valid bits.
                                                        uint8_t rxAlign,       ///< In: Defines the bit position in backData[0] for the first bit received. Default 0.
                                                        bool checkCRC          ///< In: True => The last two bytes of the response is assumed to be a CRC_A that must be validated.
                                     )
                                     {

    int status;
	// Prepare values for BitFramingReg
	byte txLastBits = validBits ? *validBits : 0;
	byte bitFraming = (rxAlign << 4) + txLastBits;		// RxAlign = BitFramingReg[6..4]. TxLastBits = BitFramingReg[2..0]

	PCD_WriteRegister(spi,CommandReg, PCD_Idle);			// Stop any active command.
	PCD_WriteRegister(spi,ComIrqReg, 0x7F);					// Clear all seven interrupt request bits
	PCD_WriteRegister(spi,FIFOLevelReg, 0x80);				// FlushBuffer = 1, FIFO initialization
	PCD_WriteRegisters(spi,FIFODataReg, sendLen, sendData);	// Write sendData to the FIFO
	PCD_WriteRegister(spi,BitFramingReg, bitFraming);		// Bit adjustments
	PCD_WriteRegister(spi,CommandReg, command);				// Execute the command
	if (command == PCD_Transceive) {
		PCD_SetRegisterBitMask(spi,BitFramingReg, 0x80);	// StartSend=1, transmission of data starts
	}

	// Wait for the command to complete.
	// In PCD_Init() we set the TAuto flag in TModeReg. This means the timer automatically starts when the PCD stops transmitting.
	// Each iteration of the do-while-loop takes 17.86μs.
	// TODO check/modify for other architectures than esp8266
	uint32_t i = 9999999;
	for (; i > 0; i--) {
		byte n;
		n = PCD_ReadRegister(spi,ComIrqReg);	// ComIrqReg[7..0] bits are: Set1 TxIRq RxIRq IdleIRq HiAlertIRq LoAlertIRq ErrIRq TimerIRq
		   if (n & waitIRq) {					// One of the interrupts that signal success has been set.
			break;
		}
		if (n & 0x01) {						// Timer interrupt - nothing received in 25ms
		return STATUS_TIMEOUT;
		}
	}
	// 35.7ms and nothing happend. Communication with the MFRC522 might be down.
	if (i == 0) {
		return STATUS_TIMEOUT;
	}

	// Stop now if any errors except collisions were detected.
	byte errorRegValue = PCD_ReadRegister(spi,ErrorReg); // ErrorReg[7..0] bits are: WrErr TempErr reserved BufferOvfl CollErr CRCErr ParityErr ProtocolErr
	if (errorRegValue & 0x13) {	 // BufferOvfl ParityErr ProtocolErr
		return STATUS_ERROR;
	}

	byte _validBits = 0;

	// If the caller wants data back, get it from the MFRC522.
	if (backData && backLen) {
		byte n = PCD_ReadRegister(spi,FIFOLevelReg);	// Number of bytes in the FIFO
		if (n > *backLen) {
			return STATUS_NO_ROOM;
		}
		*backLen = n;											// Number of bytes returned
		PCD_ReadRegisters(spi,FIFODataReg, n, backData, rxAlign);	// Get received data from FIFO
		_validBits = PCD_ReadRegister(spi,ControlReg) & 0x07;		// RxLastBits[2:0] indicates the number of valid bits in the last received byte. If this value is 000b, the whole byte is valid.
		if (validBits) {
			*validBits = _validBits;
		}
	}

	// Tell about collisions
	if (errorRegValue & 0x08) {		// CollErr
		return STATUS_COLLISION;
	}

	// Perform CRC_A validation if requested.
	if (backData && backLen && checkCRC) {
		// In this case a MIFARE Classic NAK is not OK.
		if (*backLen == 1 && _validBits == 4) {
			return STATUS_MIFARE_NACK;
		}
		// We need at least the CRC_A value and all 8 bits of the last byte must be received.
		if (*backLen < 2 || _validBits != 0) {
			return STATUS_CRC_WRONG;
		}
		// Verify CRC_A - do our own calculation and store the control in controlBuffer.
		byte controlBuffer[2];
		status = PCD_CalculateCRC(spi,&backData[0], *backLen - 2, &controlBuffer[0]);
		if (status != STATUS_OK) {
			return status;
		}
		if ((backData[*backLen - 2] != controlBuffer[0]) || (backData[*backLen - 1] != controlBuffer[1])) {
			return STATUS_CRC_WRONG;
		}
	}

	return STATUS_OK;
} // End PCD_CommunicateWithPICC()