miguelbalboa/rfid #209: Projet HighSchool serial monitor issue

Raphidu92, 2016-05-03:
Hi,
In order to my project in my final year of high school, I'm trying to make work the library with the example DumpInfo. Unfortunately, on the serial monitor, nothing is displayed, not even hexadecimal tag ID or card ID and I don't know why. Can you help me and, eventually, send me a proper version of this library to make this library work for my project ? 
Thanks.

Raphidu92, 2016-05-03:
Byt the way, I am using an Arduino Uno.
Thanks a lot.

Rotzbua, 2016-05-03:
Hi Raphidu92,
the issue tracker is only for bugs, improvements or feature requests.
Actually this library works. The serial monitor is not part of this project, it is part of the arduino ide.
For individual help please use the official arduino forum: https://forum.arduino.cc .
Best

mdxs, 2016-05-15:
Issue can be closed.

closed 2016-05-15 by miguelbalboa