miguelbalboa/rfid #212: ESP8266 bug cpu type

asetyde, 2016-05-22:
[image]

Correct this with stm32f1 in library.json

Rotzbua, 2016-05-28:
`ststm32` has nothing to do with the `esp8266`.

I added the notice that the library works with esp8266, but it do not mean that this library will grant continuously support. It is more like a bonus.

For `ststm32` or `stm32f1` I can not make a conclusion.

asetyde, 2016-05-28:
oh , i did a mistake , sorry.
Can you add compability tags to permit working ?
arduino ide go but visal studio not if platform is not indicted , (it’s a good idea )

commit e9003130 (2016-08-25): upd readme

add note for compatible ide
closes #200
closes #204
closes #205
closes #212
upd readme
closes #220
closes #36
extend gitignore with ide clion data
closed 2016-08-25 by Rotzbua